realsenseai/librealsense #11648: Can't get color image from rs pipeline.

Landon-Lamoreaux, 2023-04-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                        | D435                            | 
| Firmware Version                    |                                      | 
| Operating System & Version  |   Linux Ubuntu 20.04    | 
| Kernel Version (Linux Only)    |  4.9                                | 
| Platform                                 | NVIDIA Jetson Nano     |
| SDK Version                           |   Latest Release              | 
| Language                               |  Python                          | 
| Segment			         |  Robot                           | 

### Issue Description

Hello,
I am trying to use the realsense D435 camera for a autonomous robot. We need to get the color, and the depth images, and then align them with a fast frame rate. The problem we are having is that through the realsense pipeline we can get the depth image, but if we try to get the color image, we get the error message: "RuntimeError: Frames didn't arrive in 5000". We can get both the color and the depth images through the Ros topics though. If we try to get an image from the depth aligned to color Ros topic we only get 2-3 fps. If we can get the color image from the pipeline I would like to use the align function provided with pyrealsense2 to align the images. As far as I know there is no way to use the pyrealsense2 align to align the 2 images we get through Ros. Is there anything I should try to get the color working through the pipeline? Or any way to make the depth aligned to color Ros topic faster? Additionally, if there is any more information you need let me know.

MartyG-RealSense, 2023-04-03:
Hi @Landon-Lamoreaux  Are you using ROS1 (Kinetic, Melodic, Noetic) or ROS2, please?

If you are using ROS1, in the ROS1 wrapper low FPS problems related to depth-color alignment or pointcloud generation are a known issue when using the wrapper with Jetson boards specifically.  

If your project is able to use a 3D pointcloud as its aligned image then a ROS1 workaround is to not set align_depth to true, and to instead produce an RGB-textured pointcoud with an **rs_rgbd** launch instead of rs_camera, as described in the procedure at https://github.com/IntelRealSense/realsense-ros/issues/2646#issuecomment-1459954964

*********

In the ROS wrapper you can use Python code from within ROS as a 'node script'to subscribe to ROS topics and access their data.  A ROS1 example of a pyrealsense2 script provided by the wrapper is **show_center_depth.py**, which outputs the depth value of the pixel at the center of the camera.

https://github.com/IntelRealSense/realsense-ros/blob/ros1-legacy/realsense2_camera/scripts/show_center_depth.py

https://github.com/IntelRealSense/librealsense/issues/6367#issuecomment-626040550 provides information about accessing a ROS node script.

show_center_depth.py uses intrinsics from "/camera/depth/camera_info" topic to calculate 3D position of a pixel whos range is obtained from the depth image, received from the "/camera/depth/image_rect_raw" topic.

I assume that if align_depth was enabled then the aligned depth topics could be subscribed to with a node script.  However, it may have the low FPS issue that you have been experiencing and so may not be a suitable solution.

**********

I do not have an existing example for instead accessing a pyrealsense2 alignment from within ROS instead of using align_depth, though as a node script grants access to pyrealsense2 I would speculate that it is technically possible to achieve.

Landon-Lamoreaux, 2023-04-03:
Hello,
I am using ROS 1 Noedic, and I did build librealsense with CUDA enabled. When I have align _depth = false, I do not get the low frame rates, only when it is set to true.

I have been searching, but I haven't yet found a way to take the color image and the depth image from ROS and align them with the pyrealsense align.

Do you know a fix for not being able to get the color image through the realsense pipeline? We get the runtime error timing out at the wait_for_frames() call. We can get the depth image, but not the color image. On my other computer I can get both the color and depth frames, the rgb camera works just fine.

MartyG-RealSense, 2023-04-03:
The rs_rgbd.launch launch file should publish its pointcloud automatically to the topic **depth_image_proc** without needing to enable the pointcloud filter.

Have you tried adding **initial_reset:=true** to the end of your roslaunch instruction to reset the camera at launch to see whether that improves performance?

There was a recent case at https://github.com/IntelRealSense/realsense-ros/issues/2666 where the color stream was very slow in ROS1 at 3 FPS but the infrared stream was not.  It was therefore possible to texture the pointcloud with the infrared stream instead of color by adding to the roslaunch the instruction **pointcloud_texture_stream:=RS2_STREAM_INFRARED**

That case has some other suggestions for dealing with a 3 FPS color stream, like the one at https://github.com/IntelRealSense/realsense-ros/issues/2666#issuecomment-1490191053 for disabling RGB auto-exposure.

Landon-Lamoreaux, 2023-04-03:
The ros launch we've been using includes the initial_reset:=true.
 The color stream from the realsense pipeline doesn't have a low fps, we get the error: "RuntimeError: Frames didn't arrive in 5000" the first time wait_for_frames() is called in the function. We can get depth at a good fps through the pipeline, but nothing from the color stream. The depth and color both work at a high fps when grabbing from ROS though, the issue with that is we can't align that depth image to the color image. Hence why we are trying to get the color stream working with the realsense pipeline, so we can easily align the 2 images with the built in align function.

MartyG-RealSense, 2023-04-03:
Have you tested the SDK's pyrealsense2 example program **align_depth2color.py**

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py

Landon-Lamoreaux, 2023-04-03:
Yes, I have tried the align_depth2color.py example. That also gives the same error where line 67: "frames = pipeline.wait_for_frames()" gives the error: "RuntimeError: frames didn't arrive within 5000". We have narrowed the issue down to the color image. If we disable the color stream, it works. If the color stream is enabled, pipeline.wait_for_frames() gives that error

MartyG-RealSense, 2023-04-03:
On line 60 of align_depth2color.py, do you still receive the error if you align **color to depth** instead of depth to color by changing 'color' to 'depth'?

`align_to = rs.stream.depth`

Landon-Lamoreaux, 2023-04-03:
I will check, but I suspect that it will not fix the issue. We created a small script that just gets the depth image from the pipeline and displays it to the screen, that worked. But if we changed it to getting the color image from the pipeline and displaying it to the screen it gave the error at the wait_for_frames() call.

Landon-Lamoreaux, 2023-04-03:
I checked, the issue persists when I change the line to align_to = rs.stream.depth

Landon-Lamoreaux, 2023-04-04:
I just checked the realsense viewer with realsense-viewer. When I enable the depth image I get the depth image. When I enable the color image I get errors and no image. The error in the terminal where I opened the realsense viewer is: "ERROR [547297917184] (synthetic-stream.cpp:48) Exception was thrown during callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_".

MartyG-RealSense, 2023-04-04:
A synthetic stream error is not traceable back to a particular cause and so is equal to a 'something went wrong' message that does not indicate what went wrong, unfortunately.

As the color stream works normally in ROS, we can assume from this that the RGB sensor itself is working correctly and it is just streaming in librealsense that has the problem in this particular case.

If the location that the camera is being used in is well illuminated then a compromise may be to use the monochrome infrared Y8 image and disable the infrared emitter to remove the IR dot pattern from the Y8 image.  Then it would resemble a black and white RGB image.  It would also be unnecessary to perform alignment as the Infrared image from the left infrared sensor is already perfectly aligned and time-synced with the depth image as the depth and left infrared images originate from the same sensor.

If you require the RGB image however, does the **RuntimeError: Frames didn't arrive in 5000** error still occur if you insert a **hardware_reset()** instruction into your script to reset the camera when the script starts?  A simple Python example of code for dong so is below.

```
ctx = rs.context()
devices = ctx.query_devices()
for dev in devices:
dev.hardware_reset()
```

Landon-Lamoreaux, 2023-04-04:
Thank you for the suggestions,
We need the camera to work on sunny and cloudy days, additionally we need to be able to mask out everything that isn't white. So we need the rgb image to determine what color things are.
I tried the hardware_reset() instruction as in your example, however it didn't fix the error. I had it right before the "try:" in the align-depth2color.py program, and that gave a RuntimeError: failed to set power state. Then I moved it to before the pipeline was started and that worked until it gets to the wait_for_frames() and then we get the frame didn't arrive within 5000 error.

Do you know anything else we could try?

MartyG-RealSense, 2023-04-05:
Would it be possible to share your Python script in a comment, please?

Landon-Lamoreaux, 2023-04-05:
Here is one of the scripts we have been testing with.

```python
# First import the library.
import pyrealsense2.pyrealsense2 as rs
# Import Numpy for easy array manipulation.
import numpy as np
# Import OpenCV for easy image rendering.
import cv2 as cv
from cv_bridge import CvBridge

# Create a Tk window with the video feed.
def show_img(frame, win_name):
    cv.imshow(win_name, frame)
    k = cv.waitKey(5) & 0xFF

# Create a pipeline.
pipeline = rs.pipeline()
config = rs.config()

#config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

profile = pipeline.start(config)
try:
    while True:

        frames = pipeline.wait_for_frames()

        image = frames.get_color_frame()
        #image = frames.get_depth_frame()
        image = np.asanyarray(image.get_data())

        show_img(image, "Image")

finally:
    pipeline.stop()
```

We have been using this to test solutions to the current problem where the color image can't be grabbed by the pipeline. I suspect that the script is not the problem since we have tested this on several different scripts. The realsense viewer will also show the depth image but not the color. However I can view the yuyv color format on the viewer, I found another issue where someone had this problem but their solution did not work for me. #10070

MartyG-RealSense, 2023-04-06:
Typically instead of using **import cv2 as cv**, `import cv2` would be used instead and therefore `cv2.imshow` would be used instead of cv.imshow.

 Also, please try reversing the order of the instructions in the bracket that imshow is imported into.

`show_img('Image', image)`

Another possible issue is using 'image' twice in  `image = np.asanyarray(image.get_data())`.  You could do this instead:

```
image_frame = frames.get_color_frame()

image = np.asanyarray(image_frame.get_data())
```

Landon-Lamoreaux, 2023-04-07:
Hello,
So the code I was using is definitely not the cause of the problem. There was something up with either how it was built or the OS. I fixed the issue by getting a new fresh install of ubuntu 20.04, once I did that, I built pyrealsense2 from source and it worked perfectly. So the code I was using works just fine, something else somewhere else was breaking it.

closed 2023-04-07 by Landon-Lamoreaux

MartyG-RealSense, 2023-04-08:
It's excellent to hear that you achieved a solution.  Thanks very much for the update!