realsenseai/librealsense #11649: RGB8 Color Stream Not Working But YUYV is on Jetson Nano

Brac24, 2023-04-02:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Firmware Version                   | 5.13 and 5.14 | 
| Operating System & Version |   Linux (Ubuntu 20)  | 
| Kernel Version (Linux Only)    |  4.9                                         | 
| Platform                                 | NVIDIA Jetson |
| SDK Version                            |  2.53.0                        | 

### Issue Description
Building librealsense from source for ARM inside a Docker image running the OS available in JetPack 5.02 so it is a version of Ubuntu 20.04. I believe I have to build from source because no ARM binaries exist for Ubuntu 20.04.

The issue is that running rs-color example returns an error
[image]

Opening realsense-viewer the color stream will work if I select the YUYV color format but all other color formats give an error on synthetic-stream.cpp line 48.

A very similar issue in the past was this one https://github.com/IntelRealSense/librealsense/issues/2306
I'm not sure how to debug this. I am using the native backend. I will try the RSUSB backend but regardless if YUYV color stream already works then the other color streams should also work and just want to understand what could be causing these other color formats to not work.

The color stream works inside Docker when the environment is Ubuntu 18.04 installed from binaries.

Thank you

MartyG-RealSense, 2023-04-03:
Hi @Brac24  You are correct that the librealsense SDK should be built from source code on Jetson because there are not Arm64 packages for installing librealsense for Ubuntu 20.04 on Arm devices.

If you are building from source on Jetson and using JetPack 5.0.2 then you should either not apply the **/scripts/patch-realsense-ubuntu-L4T.sh** kernel patch script or instead build the SDK in RSUSB backend mode.  This is because JetPack 5.0.2 is not officially supported until the next SDK version after 2.53.1 (likely numbered 2.54.0).

MartyG-RealSense, 2023-04-10:
Hi @Brac24  Do you require further assistance with this case, please?  Thanks!

Brac24, 2023-04-16:
Yeah I did try both RSUSB and also not applying kernel patch but still doesn't work as expected. It's ok I will try to use an earlier JetPack that uses Ubuntu 18 for now.

closed 2023-04-16 by Brac24

MartyG-RealSense, 2023-04-16:
Thanks very much @Brac24 for the update.