realsenseai/librealsense #11654: Can not retreive the DEPTH SENSOR data

leoteo1, 2023-04-04:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515| 
| Firmware Version                   |1.5.8.1 | 
| Operating System & Version |   {Win 10 Home |                                    | 
| Platform                                 | PC |
| SDK Version                            | Intel Realsense v2.48 | 
| Language                            | C++/ Run IntelRealsenseViewer_v2.48 | 
| Segment			|  others  | 

The depth sensor is error with message:

"Backend in rs2_set_option(options:0000026648171740, option:Visual Preset,
value:5):
hr returned: HResult 0x8007001f: "A device attached to the system is not
functioning." 

Sensor succesfully connected, see the bottom image.

[image]

MartyG-RealSense, 2023-04-04:
Hi @leoteo1  Could you first try unplugging the micro-sized end of the USB cable from the base of the camera, reversing the connector's orientation and plugging it back into the L515 camera, please (USB-C cables are two-way insertion at the micro-sized end).  Does doing so resolve the error?

leoteo1, 2023-04-05:
Hi Marry, 

Yes, I tried to unplug and plug again but **nothing changed**. 
I also have tried to do similar way as the thread (https://github.com/IntelRealSense/librealsense/issues/10954).
With Reasense viiewer V2.5.3 I can not connect the L515 until I downgraded to V2.48. Currently the L515 is show up, RGB camera and motion can work without any problem. Only Depth sensor not work.

Look at the Lidar sensor I can not see red-light (it seems the IR sensor can not active).

leoteo1, 2023-04-06:
> Changing a value in rs-software-device should not affect the camera.
> 
> May I ask the following questions please?
> 
> 1. Are you using the official 1 meter USB cable supplied with the camera or a longer USB cable of your own choice? -
=> Yes, I am using the original cable from Realsense

> 2. Does the depth sensor work if you install the previous firmware version **1.5.5.0**?  You can do this by downloading the firmware .bin file for 1.5.5.0 from the link below and then going to the **More** option near the top of the Viewer's options side-panel, selecting 'Update Firmware' from its drop-down menu and navigating to the location on the computer that you downloaded the .bin file to.  Select the .bin file to start the firmware downgrade.
> 
> https://dev.intelrealsense.com/docs/firmware-releases-l500#l500-series-firmware-downloads
> 
=> It still not work (same as firmware V1.5.8)

> 3. Please use the instructions at [RealSense Viewer not working when camera connected #3381 (comment)](https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789) to check the Windows camera privacy options to confirm that camera access permissions are set to On.
=> check an the Window granted for camera access

> 4. Does changing the preset from Max Range to Short Range and then enabling depth make a difference?
>=> I did so but depth sensor still not work. Only error message show up: 
"Backend in rs2_set_option(options:0000026648171740, option:Visual Preset,
value:5):
hr returned: HResult 0x8007001f: "A device attached to the system is not
functioning" 

> If you are using the official cable supplied with the camera and 2 to 4 did not solve the problem then it is likely that there is a hardware fault in the camera that cannot be corrected. If the camera is less than a year old then you should contact the retailer that you purchased the camera from about returning it.

**Unfortunatly, my L515 guarantee is end, so is there any solution??** 

[image]

MartyG-RealSense, 2023-04-06:
Are the streams able to start and the red camera light come on if you set the **Laser Power** option to '0' in the **L500 Depth Sensor > Controls** section of the Viewer's side-panel and then enable the depth sensor?

[image]

If the red light of the camera still does not come on when the depth sensor is enabled then it is likely that there is indeed a hardware failure in the camera that cannot be fixed, unfortunately.

leoteo1, 2023-04-06:
Unfortunatly, I tried to downgrade the firmware from v1.5.8 to v1.5.5 and notthing changed. After that I downgraded again from V1.5.5 to 1.5.3, the L515 is dissapper from windows and RealsenseViewer V2.4.3 also. 

Now I can not do anything more 😂

MartyG-RealSense, 2023-04-06:
Downgrading the firmware too far will remove USB 2 support, as L515 was originally a USB3-only camera.  If the camera was detected as being on a USB 2.1 connection (which can happen if the micro-sized USB connector is the wrong way around) then it would disappear.  Re-upgrading the firmware on a USB 3 connection - which may be restored by reversing the connector - should restore USB 2 support.

leoteo1, 2023-04-06:
How can I upgrade the firmware? Currently,  RealsenseViewer v2.41 or others can not detect the L515 device so rs-update not work??

MartyG-RealSense, 2023-04-06:
L515 cameras should be updated through the RealSense Viewer instead of rs-fw-update.  This is a problem for you, as the Viewer is not detecting your camera.

If you return to Viewer 2.53.1 using the **Intel.RealSense.Viewer.exe** file from the Assets list of that version, is the camera still undetectable?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.53.1

Intel.RealSense.Viewer.exe should be launched from the folder location on your computer that it was downloaded to and not with the launch shortcut on the Windows desktop.

leoteo1, 2023-04-07:
>Intel.RealSense.Viewer.exe should be launched from the folder location on your computer that it was downloaded to and not with the launch shortcut on the Windows desktop.
**=> YES, I have tried but nothing change.**

 L515 cameras should be updated through the RealSense Viewer instead of rs-fw-update. This is a problem for you, as the Viewer is not detecting your camera. 

**=> YES, the L515 is not recognized in both RealSenseViewer.exe V2.48 or V2.53.1 so I can not do anything.**   

[image]

MartyG-RealSense, 2023-04-07:
I downgraded my L515's firmware to 1.5.3.0 and it was able to still be detected in the 2.53.1 version of the Viewer.  I was also able to re-upgrade the L515 back to 1.5.8.1.  So there is a problem with your particular L515.

If you look at the Windows **Device Manager** window, does your L515 camera appear in the 'Cameras' section or the 'Universal Serial Bus Devices' section?

[image]

leoteo1, 2023-04-08:
YES, my L515 can not be recognized by WINDOWs after downgrade firmware to V1.5.3.0 (see the bottom picture). That why I can not do anything more... Is that the old L515 hardware problem (similar to problem from this thread https://github.com/IntelRealSense/librealsense/issues/10954)?? 

My L515 info:
Bought in December 2021. 
K69571 - 100
F0244745 999NGF
Made in Thailand.

[image]

MartyG-RealSense, 2023-04-08:
Could you send me a short email at the address below please so that I can respond to you by mail about the next step.
martyx.grover@intel.com

MartyG-RealSense, 2023-05-19:
Hi @leoteo1  Was this issue resolved to your satisfaction?  If so, may the case now be closed?  Thanks!

MartyG-RealSense, 2023-05-30:
Hi @leoteo1  Do you require further assistance with this case, please?  Thanks!

leoteo1, 2023-05-31:
Hi Matry G, 

I'm happy with quick response and the way your company solve customer's problems. Thanks a lot.

MartyG-RealSense, 2023-05-31:
You are very welcome  :)   As you have received the camera, I will close this issue.  Thanks again!

closed 2023-05-31 by MartyG-RealSense