realsenseai/librealsense #11656: Calibration problem D457

lpj20, 2023-04-06:
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D457 } | 
| Firmware Version                   | (05.14.00.00) | 
| Operating System & Version |  Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.53.1 }                          | 

### Issue Description
Hi, I bought a RealSense D457 (USB 3.2) recently and prepared to calibrate the camera. I downloaded the "Librscalibrationtool" tool but when the tool was started, the default resolution is 1280*720. But in my case, I hope to calibrate the camera in 640*480 resolution. I would appreciate it if you could give me some suggestions about how to change the default resolution when I open Dynamic Calibrator v2.12.

lpj20, 2023-04-06:
> Hi @lpj20 You do not need to select a resolution when using the RealSense calibration tools. RealSense will make the appropriate adjustments automatically when you change resolution.
> 
> If you launch the 'rs-enumerate-devices' tool in calibration mode in the Ubuntu terminal with the command below then it lists the calibration intrinsics for each supported stream format and resolution on the camera that is plugged in, and you can see that the calibration values differ between resolutions.
> 
> **rs-enumerate-devices -c**

Great, @MartyG-RealSense. But I have two other questions.
The first one is that I got five coeffs results in the terminal, are that mean k1, k2, t1, t2, k3, respectively?
The second one is that the result returned from "rs-enumerate-devices -c" is different from the result I got by command "rostopic echo /camera/color/camera_info". For example, the fx, fy results from "rs-enumerate-devices -c" are 388.769 and 388.275(left pic), and the results from "rostopic" are 388.980 and 388.486(right pic). Therefore, I am wondering why there is a difference between the two results and which one is correct. Thank you very much.
[image]

MartyG-RealSense, 2023-04-06:
For the Brown-Conrady distortion model used by RealSense 400 Series cameras, the five coefficients are listed as k1, k2, p1, p2, k3.  The link below states that different distortion models have different interpretations of the five coefficients.

https://intelrealsense.github.io/librealsense/doxygen/structrs2__intrinsics.html#acc6966b80705aa25f3b6903d173d575c

ROS may report the distortion model's name as being Plumb Bob, and RealSense ROS users at https://github.com/IntelRealSense/realsense-ros/issues/709 suggest that Plumb Bob is equivalent to the Brown-Conrady distortion model reported by rs-enumerate-devices -c.

I recommend using the values provided for Brown-Conrady by rs-enumerate-devices, as it is an official tool designed specifically for RealSense cameras.

lpj20, 2023-04-07:
> For the Brown-Conrady distortion model used by RealSense 400 Series cameras, the five coefficients are listed as k1, k2, p1, p2, k3. The link below states that different distortion models have different interpretations of the five coefficients.
> 
> https://intelrealsense.github.io/librealsense/doxygen/structrs2__intrinsics.html#acc6966b80705aa25f3b6903d173d575c
> 
> ROS may report the distortion model's name as being Plumb Bob, and RealSense ROS users at [IntelRealSense/realsense-ros#709](https://github.com/IntelRealSense/realsense-ros/issues/709) suggest that Plumb Bob is equivalent to the Brown-Conrady distortion model reported by rs-enumerate-devices -c.
> 
> I recommend using the values provided for Brown-Conrady by rs-enumerate-devices, as it is an official tool designed specifically for RealSense cameras.

@MartyG-RealSense. 
① I know the Plumb Bob results seem to be equivalent to that of the Brown Conrady in distortion model (you can see in the last pic.). But there are large differences between fx,fy, ppx and ppy. These parameters seem irrelevant to the distortion model, why cause that?
② When I used the D457 camera, the Viewer shows device was D455, is that OK? Thank you!
[image]

MartyG-RealSense, 2023-04-07:
The RealSense SDK does not use the 5 coefficients when estimating fx, fy, ppx and ppy - as advised by a RealSense team member at https://github.com/IntelRealSense/librealsense/issues/1430 - and so the coefficients will indeed be irrelevant to the distortion model.

As mentioned above, if you have two sets of different values (one set from from official RealSense tools such as rs-enumerate-devices and another set from tools not specific to RealSense, such as ROS) and are in doubt about which set is correct then rely on values from official RealSense tools that can access the calibration data stored inside the camera hardware.

In regard to the D457 being detected as D455 when its mode switch is set to USB, this is correct and normal.  A D457 in USB mode has the same performance and feature-set as a D455, with the added IP65 dust and water protection provided by its casing.

lpj20, 2023-04-07:
> The RealSense SDK does not use the 5 coefficients when estimating fx, fy, ppx and ppy - as advised by a RealSense team member at #1430 - and so the coefficients will indeed be irrelevant to the [...]

@MartyG-RealSense Thank you for your patience :)

closed 2023-04-07 by lpj20