realsenseai/librealsense #11658: double free or corruption (!prev) // malloc(): memory corruption

rayprabhakarray, 2023-04-06:
Hello,

Thank you for supporting us with realsense issues. Please could you help me with the below?

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i| 
| Firmware Version                   |05.13.00.50 | 
| Operating System & Version | Linux(Ubuntu 18)  | 
| Kernel Version (Linux Only)    | 4.15.0-204-generic  | 
| Platform                                 | PC  |
| SDK Version                            |   LibRealSense v2.50.0   | 
| Language                            |  Python | 
| Segment			|  Robot        | 
|ROS distro                  | Melodic |
|ROS wrapper Version    | v2.3.2|

### Issue Description
Upon launching the realsense node ( roslaunch realsense2_camera rs_camera.launch initial_reset:=true
), rostopic such as /camera/depth/image_rect_raw, /camera/color/image_raw leads to node crash with the below error
malloc(): memory corruption (sometimes with double free or corruption (!prev))

###Logs

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
[... 89 lines cut ...]
[ INFO] [1680800368.006180621]: Motion Module was found.
[ INFO] [1680800368.006251354]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1680800368.006313923]: num_filters: 0
[ INFO] [1680800368.006346245]: Setting Dynamic reconfig parameters.
[ INFO] [1680800368.258232684]: Done Setting Dynamic reconfig parameters.
[ INFO] [1680800368.259527639]: depth stream is enabled - width: 640, height: 480, fps: 15, Format: Z16
[ INFO] [1680800368.260736734]: color stream is enabled - width: 640, height: 480, fps: 15, Format: RGB8
[ INFO] [1680800368.260809545]: setupPublishers...
[ INFO] [1680800368.269795272]: Expected frequency for depth = 15.00000
[ INFO] [1680800368.282707005]: Expected frequency for color = 15.00000
[ INFO] [1680800368.296597393]: setupStreams...
[ INFO] [1680800368.397651330]: SELECTED BASE:Depth, 0
[ INFO] [1680800368.461959580]: RealSense Node Is Up!
[ WARN] [1680800368.585991461]: 
[ WARN] [1680800370.259635545]: **Hardware Notification:Depth stream start failure,1.6808e+12,Error,Hardware Error
malloc(): memory corruption**
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

### Note:
1. depth and color streams are fine in the realsense_viewer app.
2. As per [this](https://github.com/IntelRealSense/realsense-ros/releases), seems like ROS wrapper and librealsense version match.
3. I have to run this using USB2 only despite reduced performance.

MartyG-RealSense, 2023-04-06:
Hi @rayprabhakarray  Yes, your librealsense and RealSense ROS wrapper versions are matched correctly.  The ROS wrapper also works with USB 2 so that should not be the problem, and firmware 5.13.0.50 is the correct firmware version to use with librealsense 2.50.0.

As you built librealsense from source code using the installation.md instruction page, did you then build the ROS wrapper from source code using the instructions at the link below?

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy#step-2-install-intel-realsense-ros-from-sources

If you have built librealsense from source code then you should not then build the wrapper from **packages** using the command `sudo apt-get install ros-$ROS_DISTRO-realsense2-camera`.  This is because this method installs both librealsense and the wrapper at the same time from packages, and so if you have already built librealsense from source code then it will be installing librealsense a second time from packages, leading to possible conflicts between the two SDK installations.

If you require the RealSense Viewer then building librealsense first (either from packages or source code) and the ROS wrapper separately from source code afterwards is the correct procedure.

rayprabhakarray, 2023-04-06:
Thank you for quick help, @MartyG-RealSense. Yes, I did not build the wrapper from packages using the command sudo apt-get install ros-$ROS_DISTRO-realsense2-camera. Instead I downloaded the Ros wrapper from  https://github.com/IntelRealSense/librealsense/tree/R/250, then followed instructions from https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy#step-2-install-intel-realsense-ros-from-sources to build the package. 

I do not require Realsense Viewer. Being able to query rostopics is my main concern.

MartyG-RealSense, 2023-04-06:
If you do not require RealSense Viewer then the easiest, most problem-free way to install librealsense and the 2.3.2 wrapper for Melodic is to install from packages with `sudo apt-get install ros-$ROS_DISTRO-realsense2-camera`

You can clean up the previous librealsense and wrapper installation before using apt-get install with the procedure below.

1.  Go to the 'build' folder of your librealsense source-code installation and run this CMake command to uninstall the SDK and clean up CMake but not build the SDK again.

`sudo make uninstall && make clean`

2.  Delete the ROS wrapper source-code installation simply by deleting the entire **catkin_ws/src** catkin workspace folder.

rayprabhakarray, 2023-04-06:
Thank you very much, @MartyG-RealSense . I will update you shortly.

rayprabhakarray, 2023-04-06:
@MartyG-RealSense, It seems to be fine but could you please help me understand how can I change the launch file directly? I am not able to locate it. I have some requirements because of which I need to add few lines to the launch file.

rayprabhakarray, 2023-04-06:
sorry never mind, I think I know where to find it. I will run some additional tests and update here shortly. Thakn you for your help.

rayprabhakarray, 2023-04-06:
all good, thank you very much for your help @MartyG-RealSense . Saved me some good amount of debugging. The node seems to launch and work well but throws some warning like  below 
control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
Also, although I do not need the realsense viewer app, it was still useful for debugging and playing around with the toggles. I would be also interested in a solution that could allow me to have both at the same time.

closed 2023-04-06 by rayprabhakarray

MartyG-RealSense, 2023-04-06:
You are very welcome.  It's excellent to hear that your package installation was successful.

The warning **control_transfer returned** can indicate a communication problem between the computer and camera but may be because of the slow USB2 connection.  If the topics appear to work correctly then the warnings can be safely ignored.

If you wish to use the Viewer then removing the current librealsense / wrapper installation and building them separately would be the only option on Ubuntu.  

If you have access to a Windows computer then you can easily download and run the Viewer as a separate standalone executable program to test the camera with without affecting the current successful librealsense / wrapper installation on your Ubuntu computer..

rayprabhakarray, 2023-04-06:
Thank you, I think using windows for viewer is a great idea because then it
won’t interfere with my ROS setup. I will try that.