realsenseai/librealsense #11659: set_region_of_interest throws exception for RGB camera

jenadzitsiuk, 2023-04-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.07.150| 
| Operating System & Version |   Linux (Ubuntu)   | 
| Kernel Version (Linux Only)    |  22.04                                       | 
| Platform                                 | PC  |
| SDK Version                            |  2.49                        | 
| Language                            |  C++                          | 
| Segment			|                         | 

### Issue Description

I can query the ROI from the camera, but cannot set it. Here is the code to reproduce:

```
     auto sensor = m_pipe.get_active_profile().get_device().query_sensors()[1]; 
     sensor.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 1);
     auto _roi_sensor = sensor.as<rs2::roi_sensor>();
     rs2::region_of_interest _region = _roi_sensor.get_region_of_interest();
     std::cout << "sensor roi:" << _region.min_x << "\t" << _region.min_y << "\t" << _region.max_x << "\t"
               << _region.max_y << std::endl;
     _roi_sensor.set_region_of_interest(_region);
```

Console output:
```
sensor roi:320    180     960     540
terminate called after throwing an instance of 'rs2::invalid_value_error'
  what():  hwmon command 0x75( b4 21c 140 3c0 ) failed (response -6= Invalid parameter)
Aborted (core dumped)
```

jenadzitsiuk, 2023-04-06:
Looks like this is the same issue, it was not resolved - https://github.com/IntelRealSense/librealsense/issues/6662

MartyG-RealSense, 2023-04-07:
Hi @jenadzitsiuk  The **Invalid parameter -6** error when setting the auto-exposure ROI was also reported at https://github.com/IntelRealSense/librealsense/issues/5308 and the RealSense user shared their script solution at https://github.com/IntelRealSense/librealsense/issues/5308#issuecomment-566171607

Another RealSense user shared an alternative solution at https://github.com/IntelRealSense/librealsense/issues/4363#issuecomment-508405760

jenadzitsiuk, 2023-04-07:
Thanks for your reply. Neither of the solutions above works for me, I get same exception.

I call the setting ROI right `after enable_device() `and `enable_stream() `is called.
One thing I notice is that if do `sleep(seconds(2))` before setting ROI, I have no exception. Is there some initialization state I have to check before setting ROI?

MartyG-RealSense, 2023-04-08:
Sometimes the getting or setting of an option will be more stable if it is nested within a **sensor.supports** check to see if the camera supports a particular option.

[image]

MartyG-RealSense, 2023-04-13:
Hi @jenadzitsiuk  Do you require further assistance with this case, please?  Thanks!

jenadzitsiuk, 2023-04-13:
@MartyG-RealSense yes, the check` sensor.supports(RS2_OPTION_ENABLE_AUTO_EXPOSURE)` did not change anything. I still get the same exception

MartyG-RealSense, 2023-04-28:
Hi @jenadzitsiuk  Do you have an update about this case that you can provide, please?  Thanks!

jenadzitsiuk, 2023-04-28:
Yes thank you. The issue is not reproducible with the newest firmware and 2.53.1 lib

closed 2023-04-28 by jenadzitsiuk

MartyG-RealSense, 2023-04-28:
That's excellent news.  Thanks very much for the update!