realsenseai/librealsense #11660: NodeJS installation and build guidelines is missing from the documents

bhomaidan1990, 2023-04-08:
[This link](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/nodejs/examples/nodejs-capture.js) to the [NodeJS build guidelines in the documentation](https://dev.intelrealsense.com/docs/nodejs-wrapper) does not exist!

Can you please either fix the link, add a valid read-me file, or delete the broken link from the documentation. thanks in advance.

MartyG-RealSense, 2023-04-08:
Hi @bhomaidan1990  The RealSense NodeJs wrapper is being removed from the RealSense SDK in its next version after 2.53.1, which is why the wrapper's link is  inaccessible.

I used the **Wayback Machine** internet page archiving site to find the most recently archived version of the page at the link below.

https://web.archive.org/web/20210906064205/https://github.com/IntelRealSense/librealsense/blob/master/wrappers/nodejs/README.md

bhomaidan1990, 2023-04-08:
@MartyG-RealSense Thanks

closed 2023-04-08 by bhomaidan1990