realsenseai/librealsense #11662: How to get a suitable rgb image and depth image to generate a point cloud image?

zechengli96, 2023-04-10:
Hi, first thanks so much for all developers that might helped with this. I am trying to convert a RGB image and a Depth image in python using open3d's function. I got this code online and this is the code:
These are the original files:

[image]

[image]

And here are the code:

import open3d as o3d

if __name__ == '__main__':
    # read the color and the depth image:
    rgb_image = o3d.io.read_image("./1.jpg")
    depth_image = o3d.io.read_image("./depth.png")

    # create an rgbd image object:
    rgbd_image = o3d.geometry.RGBDImage.create_from_color_and_depth(
        rgb_image, depth_image, convert_rgb_to_intensity=False)
    # use the rgbd image to create point cloud:
    pcd = o3d.geometry.PointCloud.create_from_rgbd_image(
        rgbd_image,
        o3d.camera.PinholeCameraIntrinsic(
            o3d.camera.PinholeCameraIntrinsicParameters.PrimeSenseDefault))
    # visualize:
    o3d.visualization.draw_geometries([pcd])
    
    
    Now I am trying to use this code based on Intel realsense, I think I will be able to get the rgb image just from snapshot the camera, but how can I get the depth image? It seems this so called depth image is different from the Stereo Mode's RGBD image. It is just called depth image, how can I get the image? Would be much appreciated. Delete if this isn't allowed.

MartyG-RealSense, 2023-04-10:
Hi @zechengli96  Is the Python Open3D pointcloud script at https://github.com/IntelRealSense/librealsense/issues/10894#issuecomment-1245803237 helpful to you, please?

zechengli96, 2023-04-11:
Thanks @MartyG-RealSense  checking it still, will close and post any progress. Thanks!

zechengli96, 2023-04-11:
@MartyG-RealSense 
Hi Sir or Madam, I have some basic question I think you can answer better than chatgpt. If you don't mind. I have read through the tutorial but I couldn't find what exact I want.

1) What method the Intel depth camera used, to identify an arbitrary image's boundary of an object? I know this might be a very vague question. My understanding is we are using depth detection, just like human's eye have a stable distance with each other, two camera have a constant distance between them so using similar triangle theory, we can get the distance of an object, of its boundary, but just look at this boundary thing, it has a significant depth difference compared to outside of the boundary, say if I put my face here, in the range of my face, in depth image, there will be all blue, but on the boundary, it is actually black, and then outside the black there is another layer. Do you know what was used, to actually decide this boundary that has to be black? Is just based on a certain depth differences? Or is there any machine learning algorithm? 

More clear version of the question is, let's say if I have a rgb image, and a depth image, what algorithm or code it used to generate the boundary/gap between each section? Do you know if there is a tutorial/paper that explain this?

2) A photo in 2D stereo mode, can we call it the image there as in "RGBD value"? Is there a script that we can use, to see on a boundary, what pattern this RGBD value look like? Or, the RGBD value inside and outside the boundary's comparison? I did some research and found there was a code that can just get 1 coordinates, from this one #2231, do I just have to modify the script, and get a line of boundary, to see what's really going on underneath?

3) One more thing, if I have a normal rgb image, and a depth image generated by stereo mode, is there anyway to align/combine these two and make a point cloud model? In another word, in the realsense tool, in 2D, we can get a rbg image and a depth image, on 3D, we can just directly get a point cloud image, is there a way we running a code from a vscode terminal to actually just have this 2 image and make a point cloud image out of it?

Best Regards, thank you so much.

zechengli96, 2023-04-11:
@MartyG-RealSense  Hi Marty, thank you so much for the reply, I have never seen anyone that down to earth like you.

I think what I am referring to the black edge, is not the edge on the left, is the edge around an object, like my face, it has an edge, I want to know how the depth camera decide to form that edge...

zechengli96, 2023-04-11:
Hi, @MartyG-RealSense, Thanks for your explanation. I want to post some of my understanding to help other people.

1) Honest, your answer to "occlusion" still not answers my question, my question is in short, how can the depth camera measure/recognize the boundary of an object? And by boundary/edge, I mean for instance, when we see using our eye to see our hand, there will be an edge of hand where there is, imagine there is a line out of our hand, in daily life we consider it to be very typical, but on the depth image, or point cloud image, does those lines mean anything special? Because they all have a depth value, so do they occlusion to form a gap between objects? I know this question description might sound very vague, but I hope you can understand where I was coming from. ( In the example below, I am not talking about the big shadowing on my head area, I am talking about in general, on the edge of an object, there will be a line, my question is how that line formed?)

For occlusion, I want to explain it in an easier-to-understand way:
1) When we generate a depth image, we always tend to get an overlap, or, a shadow area, let's see this image:
[image]
See I am sitting in front of the camera, and there is a shadow area on the back of my head. On the normal RGB image, we people will not see that shadowing area, the reason depth image see that is because:

1. The position of RGB camera and Depth camera is different, so Depth camera will be able to see something where blocked in RGB camera, we call this phenomenon as occlusion.
2. This will happen when there are 3 objects, object in the back, object in the middle, and camera itself. 
3. Because depth camera sees it, so, in order to combine/align depth and color camera together to form a point cloud image,  when forming the point cloud image, the depth camera will populate that part with a shadow. The area of the shadow will depend on the distance between depth camera and color camera, and also the depth. The shape of shadow will sometime depend on the order of the depth camera and color camera, if they horizontal or vertical. And sometime, both will be affected.

( I will post my finding after trying the codes.)

zechengli96, 2023-04-12:
The only edge solution works really well. Here are the step:
1) Go to https://github.com/IntelRealSense/librealsense/tree/master/examples/pointcloud
Then go to Additional Presets not in the viewer:
download and save the EdgeMapD435.json by right click the browser, save it local
2) Open the realsense viewer, close your stream, there is a little upload button on the left panel, upload, select all file option then pick this json file.
3) Then load it, you will get a very nice just edge picture.

Here are the example:
[image]
[image]

Thanks @MartyG-RealSense

zechengli96, 2023-04-12:
Hi @MartyG-RealSense when trying the final step, I encountered some challenge, hope you can help me.
So, this is the step I took:
1) Download the librealsense-master code base
2) Open it in VScode, navigate to the examples/pointcloud/rs-pointcloud.cpp file
3) Fixed all of the errors:
1. cannot locate the example.hpp, I fixed this by changing the relative path to actual path
2. cannot find the GLFW folder, and cannot find the glut lib, I fixed by installed them
3. Right now I am backing to this file rs-pointcloud.cpp, when I try to build it, it finished with errors and there is no other clear debug information, I had my camera hooked in, and I had my app stream shut, still does not work. Below is the error build info, any insight?

Executing task: C/C++: g++ build active file

Starting build...
/usr/bin/g++ -fdiagnostics-color=always -g /home/zli/Documents/librealsense-master/examples/pointcloud/rs-pointcloud.cpp -o /home/zli/Documents/librealsense-master/examples/pointcloud/rs-pointcloud
/usr/bin/ld: /tmp/cca6zvzr.o: in function `set_viewport(rect const&)':
/home/zli/Documents/librealsense-master/examples/example.hpp:135: undefined reference to `glViewport'
/usr/bin/ld: /home/zli/Documents/librealsense-master/examples/example.hpp:136: undefined reference to `glLoadIdentity'
/usr/bin/ld: /home/zli/Documents/librealsense-master/examples/example.hpp:137: undefined reference to `glMatrixMode'
/usr/bin/ld: /home/zli/Documents/librealsense-master/examples/example.hpp:138: undefined reference to `glOrtho'
/usr/bin/ld: /tmp/cca6zvzr.o: in function `draw_pointcloud(float, float, glfw_state&, rs2::points&)':
/home/zli/Documents/librealsense-master/examples/example.hpp:948: undefined reference to `glLoadIdentity'
/usr/bin/ld: /home/zli/Documents/librealsense-master/examples/example.hpp:949: undefined reference to `glPushAttrib'
/usr/bin/ld: /home/zli/Documents/librealsense-master/examples/example.hpp:951: undefined reference to `glClearColor'
/usr/bin/ld: /home/zli/Documents/librealsense-master/examples/example.hpp:952: undefined reference to `glClear'
/usr/bin/ld: /home/zli/Documents/librealsense-master/examples/example.hpp:954: undefined reference to `glMatrixMode'
/usr/bin/ld: /home/zli/Documents/librealsense-master/examples/example.hpp:955: undefined reference to `glPushMatrix'
/usr/bin/ld: /home/zli/Documents/librealsense-master/examples/example.hpp:956: undefined reference to `gluPerspective'
/usr/bin/ld: /home/zli/Documents/librealsense-master/examples/example.hpp:958: undefined reference to `glMatrixMode'
/usr/bin/ld: /home/zli/Documents/librealsense-master/examples/example.hpp:959: undefined reference to `glPushMatrix'
/usr/bin/ld: /home/zli/Documents/librealsense-master/examples/example.hpp:960: undefined reference to `gluLookAt'
/usr/bin/ld: /home/zli/Documents/librealsense-master/examples/example.hpp:962: undefined reference to `glTranslatef'
/usr/bin/ld: /home/zli/Documents/librealsense-master/examples/example.hpp:963: undefined reference to `glRotated'
/usr/bin/ld: /home/zli/Documents/librealsense-master/examples/example.hpp:964: undefined reference to `glRotated'
/usr/bin/ld: /home/zli/Documents/librealsense-master/examples/example.hpp:965: undefined reference to `glTranslatef'
/usr/bin/ld: /home/zli/Documents/librealsense-master/examples/example.hpp:967: undefined reference to `glPointSize'
/usr/bin/ld: /home/zli/Documents/librealsense-master/examples/example.hpp:968: undefined reference to `glEnable'
/usr/bin/ld: /home/zli/Documents/librealsense-master/examples/example.hpp:969: undefined reference to `glEnable'
/usr/bin/ld: /home/zli/Documents/librealsense-master/examples/example.hpp:970: undefined reference to `glBindTexture'
/usr/bin/ld: /home/zli/Documents/librealsense-master/examples/example.hpp:972: undefined reference to `glTexParameterfv'
/usr/bin/ld: /home/zli/Documents/librealsense-master/examples/example.hpp:973: undefined reference to `glTexParameteri'
/usr/bin/ld: /home/zli/Documents/librealsense-master/examples/example.hpp:974: undefined reference to `glTexParameteri'
/usr/bin/ld: /home/zli/Documents/librealsense-master/examples/example.hpp:975: undefined reference to `glBegin'
/usr/bin/ld: /home/zli/Documents/librealsense-master/examples/example.hpp:986: undefined reference to `glVertex3fv'
/usr/bin/ld: /home/zli/Documents/librealsense-master/examples/example.hpp:987: undefined reference to `glTexCoord2fv'
/usr/bin/ld: /home/zli/Documents/librealsense-master/examples/example.hpp:992: undefined reference to `glEnd'
/usr/bin/ld: /home/zli/Documents/librealsense-master/examples/example.hpp:993: undefined reference to `glPopMatrix'
/usr/bin/ld: /home/zli/Documents/librealsense-master/examples/example.hpp:994: undefined reference to `glMatrixMode'
/usr/bin/ld: /home/zli/Documents/librealsense-master/examples/example.hpp:995: undefined reference to `glPopMatrix'
/usr/bin/ld: /home/zli/Documents/librealsense-master/examples/example.hpp:996: undefined reference to `glPopAttrib'
/usr/bin/ld: /tmp/cca6zvzr.o: in function `draw_pointcloud_wrt_world(float, float, glfw_state&, rs2::points&, rs2_pose&, float*, std::vector<rs2_vector, std::allocator<rs2_vector> >&)':
/home/zli/Documents/librealsense-master/examples/example.hpp:1014: undefined reference to `glLoadIdentity'
/usr/bin/ld: /home/zli/Documents/librealsense-master/examples/example.hpp:1015: undefined reference to `glPushAttrib'
[... cut ...]
 *  Terminal will be reused by tasks, press any key to close it.

Seems there are lots of undefined reference, I have tried multiple ways of fixing it, including:

 Changing the cmake.txt file
Check the usr/ to find the build libreal package (which I cannot find)
run the file using command like g++ -o runfile rs-pointcloud.cpp -lrealsense2 

Do I just need to instead of using zip file and all manually, but do a way of git clone and compile everything? Is there a standard step I can follow?

@MartyG-RealSense  Thank you so much for the help!

MartyG-RealSense, 2023-04-12:
There is not information available on how the RealSense depth algorithm identifies edges.  Some aspects of RealSense technology are 'closed source' and not publicly available in order for Intel to protect the confidential details of them.  As mentioned above though, you can isolate the edges on the depth image using a preset like you have done.

**********

RealSense SDK example programs are compiled differently from how programs are usually built when created by a RealSense user.  So if the code is directly copied from an example program's cpp into your own program then it will likely not work because it will be missing vital linkages to the librealsense library.  This can be corrected in the CMakeLists.txt file of a project.  

The **Getting Started** project at the link below demonstrates this.  It is the same script as the SDK's **rs-hello-realsense** example program but has a more complex CMakeLists.txt file with the linkages necessary to run the project standalone from the SDK.

https://github.com/zivsha/librealsense/tree/getting_started_example/examples/getting-started

https://github.com/zivsha/librealsense/blob/getting_started_example/examples/getting-started/CMakeLists.txt

zechengli96, 2023-04-12:
Hi @MartyG-RealSense , I am still having issue.
I am able to run the getting-started program myself and make it work.
But when I go to the point cloud example, when I do 
mkdir build
cd build 
cmake ../

Right now is all working 
when I do make -j4
the terminal has no reaction

Is it because the cmake file is different? I used the make file on the getting started to this point cloud one and this is what I get:

make -j4
[ 50%] Building CXX object CMakeFiles/rs-pointcloud.dir/rs-pointcloud.cpp.o
[100%] Linking CXX executable rs-pointcloud
/usr/bin/ld: CMakeFiles/rs-pointcloud.dir/rs-pointcloud.cpp.o: in function `set_viewport(rect const&)':
rs-pointcloud.cpp:(.text+0xdbd): undefined reference to `glViewport'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0xdc2): undefined reference to `glLoadIdentity'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0xdcc): undefined reference to `glMatrixMode'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0xe1f): undefined reference to `glOrtho'
/usr/bin/ld: CMakeFiles/rs-pointcloud.dir/rs-pointcloud.cpp.o: in function `draw_pointcloud(float, float, glfw_state&, rs2::points&)':
rs-pointcloud.cpp:(.text+0xe6c): undefined reference to `glLoadIdentity'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0xe76): undefined reference to `glPushAttrib'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0xe9d): undefined reference to `glClearColor'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0xea7): undefined reference to `glClear'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0xeb1): undefined reference to `glMatrixMode'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0xeb6): undefined reference to `glPushMatrix'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0xef1): undefined reference to `gluPerspective'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0xefb): undefined reference to `glMatrixMode'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0xf00): undefined reference to `glPushMatrix'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0xf4f): undefined reference to `gluLookAt'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0xf8a): undefined reference to `glTranslatef'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0xfb0): undefined reference to `glRotated'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0xfd5): undefined reference to `glRotated'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0xff0): undefined reference to `glTranslatef'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0x100e): undefined reference to `glPointSize'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0x1018): undefined reference to `glEnable'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0x1022): undefined reference to `glEnable'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0x103e): undefined reference to `glBindTexture'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0x1088): undefined reference to `glTexParameterfv'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0x109c): undefined reference to `glTexParameteri'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0x10b0): undefined reference to `glTexParameteri'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0x10ba): undefined reference to `glBegin'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0x1147): undefined reference to `glVertex3fv'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0x116b): undefined reference to `glTexCoord2fv'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0x1194): undefined reference to `glEnd'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0x1199): undefined reference to `glPopMatrix'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0x11a3): undefined reference to `glMatrixMode'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0x11a8): undefined reference to `glPopMatrix'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0x11ad): undefined reference to `glPopAttrib'
/usr/bin/ld: CMakeFiles/rs-pointcloud.dir/rs-pointcloud.cpp.o: in function `draw_pointcloud_wrt_world(float, float, glfw_state&, rs2::points&, rs2_pose&, float*, std::vector<rs2_vector, std::allocator<rs2_vector> >&)':
rs-pointcloud.cpp:(.text+0x1596): undefined reference to `glLoadIdentity'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0x15a0): undefined reference to `glPushAttrib'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0x15c7): undefined reference to `glClearColor'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0x15d1): undefined reference to `glClear'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0x15db): undefined reference to `glMatrixMode'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0x15e0): undefined reference to `glPushMatrix'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0x1621): undefined reference to `gluPerspective'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0x162b): undefined reference to `glMatrixMode'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0x1630): undefined reference to `glPushMatrix'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0x166a): undefined reference to `glTranslatef'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0x1693): undefined reference to `glRotated'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0x16bb): undefined reference to `glRotated'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0x16d6): undefined reference to `glTranslatef'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0x16e0): undefined reference to `glEnable'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0x16ef): undefined reference to `glLineWidth'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0x16f9): undefined reference to `glBegin'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0x1763): undefined reference to `glColor3f'
/usr/bin/ld: rs-pointcloud.cpp:(.text+0x178a): undefined reference to `glVertex3f'
[... 50 lines cut ...]
/usr/bin/ld: rs-pointcloud.cpp:(.text._ZN6windowC2EiiPKc[_ZN6windowC5EiiPKc]+0x1c0): undefined reference to `glfwSetWindowUserPointer'
[... cut ...]
/usr/bin/ld: rs-pointcloud.cpp:(.text._ZN6windowcvbEv[_ZN6windowcvbEv]+0xd2): undefined reference to `glfwGetWindowSize'
/usr/bin/ld: rs-pointcloud.cpp:(.text._ZN6windowcvbEv[_ZN6windowcvbEv]+0x12b): undefined reference to `glOrtho'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/rs-pointcloud.dir/build.make:98: rs-pointcloud] Error 1
make[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/rs-pointcloud.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

make -j4 is working but still having issue, any other insight, thank u so much. @MartyG-RealSense

zechengli96, 2023-04-12:
@MartyG-RealSense  Is there anyway to get a runnable cmakefile.txt file?

MartyG-RealSense, 2023-04-13:
Advice is provided at https://github.com/IntelRealSense/librealsense/issues/4277#issuecomment-505242118 about linking a project to the librealsense libraies using CMake.

MartyG-RealSense, 2023-04-14:
If you have exported a .ply pointcloud file from the 3D pointcloud mode of the Viewer then you can import the ply into a 3D modelling tool to do further processing on it.  A RealSense-compatible tool to do this is the free open-source **MeshLab** program.

https://www.meshlab.net/

The link below has a guide for importing a RealSense ply into MeshLab.

https://www.andreasjakl.com/capturing-3d-point-cloud-intel-realsense-converting-mesh-meshlab/

The link below advises how to display the coordinates of a specific point in MeshLab.

https://stackoverflow.com/questions/53178374/meshlab-how-do-i-extract-x-y-z-coordinates-of-points

MartyG-RealSense, 2023-04-20:
Hi @zechengli96   Do you require further assistance with this case, please?  Thanks!

zechengli96, 2023-04-25:
@MartyG-RealSense  Hi, I think we can close this issue. I am currently all set. Thanks.

closed 2023-04-25 by zechengli96