realsenseai/librealsense #11663: I want to output the FPS value in the D435i camera config settings(C#)

KITAE-KIM-MOBYUS, 2023-04-10:
I want to output the FPS value in the D435i camera config settings with consoleWriteLine(); What should I do?

[image]

The part circled in red is the framerate part.

MartyG-RealSense, 2023-04-10:
Hi @KITAE-KIM-MOBYUS  https://github.com/IntelRealSense/librealsense/issues/11111 is the best available example of C# code for printing the current actual FPS using **consoleWriteLine();**

KITAE-KIM-MOBYUS, 2023-04-12:
@MartyG-RealSense  
[image]

If I change it in C# as above, will the FPS be displayed in the command window in a changed state?

KITAE-KIM-MOBYUS, 2023-04-12:
@MartyG-RealSense  How can I check the set FPS in real time?

MartyG-RealSense, 2023-04-12:
An example of C# code for doing so is referenced above at https://github.com/IntelRealSense/librealsense/issues/11663#issuecomment-1501489014

closed 2023-04-25 by MartyG-RealSense