realsenseai/librealsense #11667: Latency for first D455 plugged into USB is highly variable.

edgex004, 2023-04-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.14.00.00 | 
| Operating System & Version |   Ubuntu 22  | 
| Kernel Version (Linux Only)    |  5.15.104 | 
| Platform                                 | PC |
| SDK Version                            |   v2.53.1  | 
| Language                            |  C++/ROS2  | 
| Segment			|  Robot | 

### Issue Description
My group is having an issue where the **first** D455 camera plugged into a system has extremely variable latency values. We have connected up to 4 D455 cameras to the system. The first camera will exhibit large +-20ms fluctuations in the latency values for both RGB and Depth images. Cameras 2-4 will always have more steady latency values: less than 1ms fluctuations generally, but with large +-20ms spikes occurring rarely. We have already filed some of our findings using ROS timestamps [here](https://github.com/IntelRealSense/realsense-ros/issues/2686).

We have begun investigating librealsense as the potential source for these fluctuations. When fluctuations occur, it is not uncommon for depth images to exhibit negative latency values. Average latency for our depth images is ~9ms. When a -20ms fluctuation occurs, this leads to a -10ms latency value being common. The latency fluctuations seem to be the same between RGB and depth, and do not scale. RGB will have latency of 66+-20ms and depth will see latency of 9+-20ms. Due to the fact that negative latency is being calculated, we suspect the root cause to be conversion between the RS clock and the host clock.

Here is a visual comparison of the latency calculations:

[image]

The data for these plots was collected from parallel streams. The first two plots are of sensors with relatively stable latency calculations. The third plot is of the first camera to be connected via USB. **The first D455 camera connected always exhibits this highly variable latency.** As you can see, even plots 1 and 2 experience "spikes" which we would like to investigate, but our largest concern for now comes from the third plot.

Latency was calculated using frame metadata provided by librealsense:
`Latency = time_of_arrival - frame_timestamp`

Here are some more detailed plots taken when only a single D455 is connected along with the source metadata:
[image]
[image].png…]()
[d455r_rgb_metadata.txt](https://github.com/IntelRealSense/librealsense/files/11192709/d455r_rgb_metadata.txt)
[d455r_depth_metadata.txt](https://github.com/IntelRealSense/librealsense/files/11192712/d455r_depth_metadata.txt)

This data was collected by running one realsense-ros node which saves each frame's metadata to a file (based on [this node](https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/realsense2_camera/tools/frame_latency/frame_latency.cpp)).

MartyG-RealSense, 2023-04-11:
hI @edgex004  Are all four D455 cameras plugged in before the computer is started up or are you inserting them after the computer has booted? 

If the above data comes from RealSense ROS, is there still latency if you reset the first camera at launch by adding **initial_reset:=true** to its launch instruction?

edgex004, 2023-04-11:
Is there anything we can do to help you reproduce on your end? We have seen the same behavior on multiple computers and multiple cameras, so it seems like it should be reproducible.

MartyG-RealSense, 2023-04-12:
I have made a suggestion about the D455 latency at https://github.com/IntelRealSense/realsense-ros/issues/2686#issuecomment-1504867495 that you can try first.

Chris45215, 2023-04-13:
I'm taking a shot in the dark here, but are the cameras always plugged into the same USB ports? If so, what happens if the first/problematic camera is never plugged in, and that port is left empty? Does the issue transfer to the next camera, or does the issue disappear?

I ask because many computers let several USB ports share a USB controller - and this can cause inconsistent performance. And few motherboard manufacturers make their USB controller setup clear. If this turns out to be the case, one solution is to use pci-e USB expansion cards. If you need to do that, I suggest getting one of the more expensive variants which has a controller for each port. Otherwise, industrial-focused computer makers often have a dedicated USB controller for each USB3 port.
But, I'm taking a shot in the dark and I hope this potential cause is easy to test.

edgex004, 2023-04-13:
@Chris45215 Unfortunately that isn't the case here from what we can tell. We had a similar idea at first, so we tried all different port combinations. The problem always presents itself in the **first** port plugged in _temporally_.

So we have been able to cause the problem on any port we choose simply by plugging the camera into that port first and then plugging other cameras into the other ports.

This rule about first camera always failing is definitely reproducible when the USB ports are plugged in while the computer is running. I am not sure if this order holds up when the USB ports are plugged into a cold system and then powered on. We definitely still have a camera failing in that case, but I'm not sure if it is dependent on temporal plug-in order or if it was based on the first port to be recognized by the kernel at boot. We could probably suss that out by checking the dmesg logs at boot.

For now, we have a workaround, but it isn't sustainable. We only use 2 D455 cameras on our system. To achieve 2 "good" cameras, we can use 3 D455 cameras: plug in a sacrificial camera first which will be plagued by the wild latency bug, then plug in two cameras that need to be "good". Unplug the sacrificial camera, and the remaining two continue to exhibit relatively stable latency numbers.

MartyG-RealSense, 2023-04-18:
Thanks very much for the update.  I do not have an Ubuntu 22.04 system to test with, so filing a Zendesk ticket to Intel would be a very good idea.

MartyG-RealSense, 2023-04-30:
Hi @edgex004  Did you file a ticket through the Intel ticketing system, please?  Thanks!

closed 2023-05-12 by MartyG-RealSense