realsenseai/librealsense #11672: Cannot play recorded bag in Unity because the "file is in use" / locked by the record process which should be stopped

cristianbercu, 2023-04-12:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   {Win (8.1/10)  | 
| Platform                                 | PC |
| SDK Version                            |  { 2.53.1 }                          | 
| Language                            |  {C#/unity }                          | 
| Segment			|  {others }                          | 

### Issue Description
I am trying to record a 5 seconds bag file, stop and play the recorded file. Unfortunately, the recording cannot be played because the "file is still in use" by the pipeline.

I set the recording profile with this:

```c#
    public void SetRecordProfile(string path)
    {
        Logger.LogGlobal("Setting sensor to Record mode");
        rsDevice.DeviceConfiguration.mode = RsConfiguration.Mode.Record;
        rsDevice.DeviceConfiguration.RecordPath = path;
    }
```

I then enable the gameObject with the sensor which calls my OnEnable() method defined as follows (similar to the demo example):

```c#
    void OnEnable()
    {
        m_pipeline = new Pipeline();

        using (var cfg = DeviceConfiguration.ToPipelineConfig())
            try
            {
                ActiveProfile = m_pipeline.Start(cfg);
            }
            catch (Exception ex)
            {
                Debug.LogError($"{ex.Message}");
                onStreamingError?.Invoke();
            }

        DeviceConfiguration.Profiles = ActiveProfile.Streams.Select(RsVideoStreamRequest.FromProfile).ToArray();

        if (processMode == ProcessMode.Multithread)
        {
            stopEvent.Reset();
            worker = new Thread(WaitForFrames);
            worker.Name = "Intel Sensor Frame Announcer";
            worker.IsBackground = true;
            worker.Start();
        }

        StartCoroutine(WaitAndStart());
    }
```

Then I stop the recording by disabling the gameobject which calls my OnDisable() method defined as follows:

```c#
    public void OnDisable()
    {
        OnNewSample = null;
        // OnNewSampleSet = null;

        if (worker != null)
        {
            stopEvent.Set();
            worker.Join();
        }

        if (Streaming && OnStop != null)
            OnStop();

        if (ActiveProfile != null)
        {
            ActiveProfile.Dispose();
            ActiveProfile = null;
        }

        if (m_pipeline != null)
        {
            // if (Streaming)
            m_pipeline.Stop();
            m_pipeline.Dispose();
            m_pipeline = null;
        }

        Streaming = false;
    }
 ```
 
 Then I set the Playback profile as follows:
 
 ```c#
     public void SetPlaybackProfile(string path)
    {
        rsDevice.DeviceConfiguration.mode = RsConfiguration.Mode.Playback;
        rsDevice.DeviceConfiguration.PlaybackFile = path;
    }
 ```
And finally, re-enable the RsDevice gameObject which calls `OnEnable()` to start playing the recording. Unfortunately, starting to play the recording fails because the file is in use. After a random amount of seconds i.e more than 10 seconds it works because I assume the pipeline is finally disposed. Why is this happening, how can I kill the recording process faster/properly? Should I not be reusing the m_pipeline?

MartyG-RealSense, 2023-04-13:
Hi @cristianbercu  Which stream types are you recording, please?  If your RealSense camera model has an IMU and you are recording Motion Module (IMU) data, there was a past issue in Unity at https://github.com/IntelRealSense/librealsense/issues/5993 where closure of the pipeline could be delayed for up to 20 seconds after streaming was stopped if the Motion Module was active.

cristianbercu, 2023-04-13:
I am recording `Depth` and `Color` streams using a D415 device.
[image]

MartyG-RealSense, 2023-04-14:
Are you building an .exe executable of your project and then closing Unity (to make sure that the camera is not currently being accessed) and launching the .exe outside of Unity?

With my own Unity projects, I use the **File > Build** option instead of 'Build and Run' to build the project in a chosen folder but not run it automatically after building has completed.  I then exit Unity and launch the .exe of the created build from my project folder.

MartyG-RealSense, 2023-04-20:
Hi @cristianbercu   Do you require further assistance with this case, please?  Thanks!

cristianbercu, 2023-04-24:
Hello @MartyG-RealSense , thank you for your assistance so far. I was using a normal Build option.
I have isolated the problem and have more findings. Issuing commands to the SDK in asynchronous calls such as coroutines makes the SDK lock the recorded file for an indefinite amount of time. Having the commands sent in sync to the SDK seems to make it work. Does the SDK have a callback to announce when the SDK finished writing to the recorded `.bag` file and has released the lock on the file?

MartyG-RealSense, 2023-04-24:
It's great to hear that you made significant progress!

The SDK's playback and record documentation indicates that callbacks are only involved in playback rather than record.

https://intelrealsense.github.io/librealsense/doxygen/rs__record__playback_8h.html#ae0fc50d7fb113c1e8a56d58f3b7bf7ce

As far as I am aware there is not a mechanism in the SDK to confirm when recording of a bag has completed.  A pipeline.stop() instruction should close the pipeline and therefore also close the recording of a bag.  Not closing the recording could lead to an incomplete recording that cannot be played back.

MartyG-RealSense, 2023-04-30:
Hi @cristianbercu  Do you have an update about this case that you can provide, please?  Thanks!

cristianbercu, 2023-05-01:
Hi Marty, sure. I have noticed that slower machines lock the recorded file for longer periods of time. I would guess that the locked time is determined by the hardware configuration such as the speed of I/O operations on the HDD, amount of RAM, USB connection type, and so on. This tells me that I should wait for the file to be unlocked by periodically ping it to see if it was unlocked if I want my solution to work on most PC's out there. It would be great if you can fact-check my findings, I'd like to make sure I am using the sensor accordingly. 🙂

MartyG-RealSense, 2023-05-01:
Computer specification can have a bearing on performance.  For example, if a computer has a storage drive with a slow access speed then it will take longer to read data from the file.  I believe that the drive speed will have a greater impact on playback than the memory or CPU.  USB should not be a factor unless the file is stored on a USB stick, in which case USB 2 will have a slower data transfer speed than USB 3.

cristianbercu, 2023-05-08:
Thanks for all your assistance!

closed 2023-05-08 by cristianbercu