realsenseai/librealsense #11674: Intel Realsense Unity

willyum00, 2023-04-12:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions , suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435I | 
| Firmware Version                   | 05.14.00.00 | 
| Operating System & Version |   Win 11  | 
| Platform                                 | PC  |
| SDK Version                            |  2.0  (v2.53.1)   | 
| Language                            |  unity      | 

Hi again Marty, 

I'm currently trying to set up the Realsense to be used with Unity. I downloaded the Unitypackage from the most recent SDK release. In the set up guide, it says to go Assets > Scenes > Start Here and this menu will show up: 
[image]

However, this is not the case for me. There is nothing under scenes except the example scene (which on another note, does not open). I'm not sure if this is something wrong with Unity or some other issue. 
[image]

Thanks

MartyG-RealSense, 2023-04-13:
Hi @willyum00  It looks as though you have successfully installed the RealSense Unity wrapper into your project.  Once the StartHere scene has loaded and your RealSense camera is plugged in, you should left-click on the 'play' triangle button at the top of the Unity window to run the scene.  The blue and white menu screen should then appear.

willyum00, 2023-04-14:
Thanks, that was the issue. 
Now, how can I set up a Point Cloud for example without using the Start Here sample?

willyum00, 2023-04-14:
Do I still need to do this step?  From https://github.com/IntelRealSense/librealsense/tree/master/wrappers/unity#prefabs. 
[image]
If so, I'm not sure how to carry it out. 
[image]

MartyG-RealSense, 2023-04-14:
The **Start Here** scene is essentially a list of launch links for individual example programs that are stored in the **Samples** folder that is next to the StartHere scene file in Assets > RealSenseSDK2.0 > Scenes

In the Samples folder, double-leftclick on the first of the two **Pointcloud...** scene files.  The full name of the scene file is **PointCloudDepthAndColor.unity**.  You can then see how the pointcloud project is structured in the Unity edtior.

In regard to the Getting Started instructions that you quoted, you do not actually need to use those.  You can just visit the SDK **Releases** page and look in the 'Assets' file list of the SDK version that you are using for the RealSense Unity package file **Intel.RealSense.unitypackage**

https://github.com/IntelRealSense/librealsense/releases

[image]

Once Intel.RealSense.unitypackage is downloaded to your computer, open a new project in Unity, go to the menu option **Assets > Import Package > Custom Package** and navigate to the folder location on your computer that you downloaded it to.  Select the file and you can then import the RealSense Unity wrapper into Unity without having to compile them with VS.

willyum00, 2023-04-19:
Hi Marty, so I've imported everything into Unity. If I want to set up a scene (not sure if that's correct terminology) with a point cloud, do I do so in the hierarchy? Where do I go to create a point cloud (without using start here?) 
[image]

MartyG-RealSense, 2023-04-19:
The simplest approach may be to open the **PointCloudDepthAndColor.unity** scene file in the `Assets > RealSenseSDK2.0 > Scenes` folder so that you have a pre-made pointcloud scene, and then build your project around that.

If you wish to replicate this project in your own existing scene then you could follow the procedure below.

1.  Create two empty GameObjects called **RsDevice** and **RsProcessingPipe**.  You create an empty GameObject by going to the menu option **GameObject > Create Empty**.

2.  Select the **RsDevice** empty GameObject that you created in the left-hand Hierarchy panel to bring up its details, then click on the 'Add Component' button in the Inspector on the right-hand side of the screen and type **rsdevice** into the search bar at the top of the small box that pops up.  This will cause the RsDevice component to be listed in the box.  Left-click on it to add the component to your RsDevice object.

3.  Select the **RsProcessingPipe** empty GameObject in the left-hand Hierarchy.  Again click on Add Component, and this time type **RsProcessingPipe** into the search box and add this component to the object.

MartyG-RealSense, 2023-04-25:
hi @willyum00  Do you require further assistance with this case, please?  Thanks!

willyum00, 2023-04-26:
Hi Marty, still working on it, will let you know if I run into more bugs. 

I followed the specific steps that you mentioned but it didn't seem to work, I think there were a few more steps I needed to do.

willyum00, 2023-04-26:
Is there a range of effectiveness for the Realsense? I've noticed that when I put the realsense close to an object, the object will disappear and only shows up if I place the realsense at least half a foot away. 
[image]

MartyG-RealSense, 2023-04-26:
The minimum depth sensing range varies between RealSense camera models but for the D435i model it should be about 0.1 meters, which converts to around half a foot, as you said.

In regard to maximum depth sensing range, whilst the camera can see up to 10 meters, error increases linearly over distance.  So beyond 3 meters, error in the measurements starts to become noticable.

MartyG-RealSense, 2023-05-02:
Hi @willyum00  Do you require further assistance with this case, please?  Thanks!

willyum00, 2023-05-03:
I had some questions about using the SDK to connect multiple realsenses, can I still use unity to display a point cloud generated from say 4 realsense cameras?

MartyG-RealSense, 2023-05-03:
The information at https://github.com/IntelRealSense/librealsense/issues/9481#issuecomment-886021966 regarding using multiple cameras in the RealSense Unity wrapper may be helpful.

closed 2023-05-14 by MartyG-RealSense