realsenseai/librealsense #11675: Realsense camera not detected on boot

HEyeRyan, 2023-04-13:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (Recent (updated from internet) | 
| Operating System & Version | Linux (Ubuntu 14/16/17) | 
| Kernel Version (Linux Only)    |  (unsure)                                         | 
| Platform                                 | ROG tablet  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python }                          | 
| Segment			|  {others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Ubuntu custom software (made in house by us)
Upon power up with camera connected via a usb3.0 to usb c adapter, the camera doesnt detect.

Unplug and plug back in, it suddenly detects.

I have tried with a 20m optical USB A cable, same thing. until you unplug and replug.

Tried common software fixes such as turning off usbcore.autosuspend

Anyone got any ideas? hardware needs to stay the same ideally.
(it does work with this hardware, but unplugging and plugging it back in isnt viable as its a commercial product)

Kind regards
Ryan

MartyG-RealSense, 2023-04-13:
Hi @HEyeRyan  If you are using a 20 meter cable, can you confirm please if it is one with **active repeater** components in it to boost the signal?  Otherwise, the camera connection will be unreliable over a USB cable of that length.  

The company **Newnex** are a supplier of industrial-grade USB equipment, including active repeater cables, that is validated for use with RealSense cameras.

https://www.newnex.com/realsense-3d-camera-connectivity.php

https://www.youtube.com/watch?v=GLQgR1jT04M

If USB **C to C** cables are used instead of A to C (Type-C) then these can also be unreliable.

If you are writing your own RealSense scripting then you could also try adding a **hardware_reset()** instruction to the script to reset the camera when the script is launched instead of resetting with a physical unplug-replug of the USB cable.  An example of Python code for hardware_reset() is below.

```
ctx = rs.context()
devices = ctx.query_devices()
for dev in devices:
dev.hardware_reset()
```

HEyeRyan, 2023-04-13:
Hi @MartyG-RealSense , it is a fibre optic cable, not active repeater, it has worked well for us for a few months, but testing it in real conditions is when we found this issue.

We are running (starting from camera end to tablet)
Camera **>** USB C - USB A 5m cable **>** USB A to USB A 20m Fibre optic cable **>** USB A - USB C 50cm adapter **>** Tablet

Our software team have also tried the hardware_reset() function

MartyG-RealSense, 2023-04-13:
You could also test building the RealSense SDK (librealsense) from source code with the **RSUSB backend = true** method if you are using unusual computing hardware or a custom version of Ubuntu.  An RSUSB build of the SDK bypasses the Linux kernel and so is not dependent on Linux versions or kernel versions and does not require kernel patching.  RSUSB is useful for resolving problems with 'exotic' computing hardware / software configurations that differ from a typical consumer PC system.

HEyeRyan, 2023-04-13:
Hi, this is being carried over to (hopefully) tomorrow, will keep updated when I have an answer whether or not this has worked.

Our 5m cable has the locking screws however looks slightly different to that, but is the same principle. 

we have done the boot tests stationary on the tractors and in the office, same issue on 4 systems all with their own set of hardware

Kind regards
ryan

closed 2023-05-02 by MartyG-RealSense