realsenseai/librealsense #11678: Why does pyrealsene2 parse more images from. bag files than rosbag or rs-convert.exe

twy123456789, 2023-04-14:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D415 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (11)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {opencv/pcl/python }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
When I used ROSBAG and RS Convert to parse. Bag files, I found that there were nearly 10000 frames missing from over 20 minutes of video. However, when saving images with Pyralsense 2, there were over 40000 frames, but still fewer than the end start frame displayed in the Intel realsense viewer. How can I save all the images

MartyG-RealSense, 2023-04-14:
Hi @twy123456789  A RealSense team member at https://github.com/IntelRealSense/librealsense/issues/3500#issuecomment-473624881 offers advice on why some bag frames may not be extracted and converted by rs-convert.

In that discussion, another RealSense team member recommended a non-RealSense Python script at https://gist.github.com/wngreene/835cda68ddd9c5416defce876a4d7dd9 for extracting all frames from a **rosbag**.  

If you recorded a rosbag in ROS then there will be differences between that bag and one recorded in the RealSense SDK, as the SDK has some differences in its bag format, as described at https://github.com/IntelRealSense/librealsense/issues/3020#issuecomment-452321766

twy123456789, 2023-04-15:
Thank you for your answer, but I did record the. bag file from the Intel realsense viewer. When using ROSBAG to view the. bag information, the frame count was much lower, nearly 10000 frames, but in fact, there should be 47000 frames. When using the ROSBAG package to extract images, only the number of images recorded in the bag file can be extracted. The problem I want to solve is that when using Pyralsense2 to extract images, there are still nearly one to two thousand frames less than the images recorded in the viewer, only 45000 frames。

MartyG-RealSense, 2023-04-15:
There is always some frame loss during extraction, unfortunately.  I have never seen a case where the count of the extracted frames matched the number of frames in the bag file.  In the best case scenario you can get close to the ideal target number but not perfectly obtain every frame.  There is not a solution where you can extract every frame to image though.

twy123456789, 2023-04-16:
Okay，thanks a lot.

MartyG-RealSense, 2023-04-21:
Hi @twy123456789  Do you require further assistance with this case, please?  Thanks!

twy123456789, 2023-04-25:
No more questions, thank you.

MartyG-RealSense, 2023-04-25:
Thanks very much for the update.  As you do not require further assistance, I will close this case.  Thanks again!

closed 2023-04-25 by MartyG-RealSense