realsenseai/librealsense #11679: How to display D435 Depth camera video in C# Window Forms?

KITAE-KIM-MOBYUS, 2023-04-14:
How to display D435 Depth camera video in C# Window Forms? The tutorial seems to be only for WPF.

MartyG-RealSense, 2023-04-14:
Hi @KITAE-KIM-MOBYUS  I hope the discussion at https://github.com/IntelRealSense/librealsense/issues/3413 about displaying depth with a C# Windows Form will be helpful.

KITAE-KIM-MOBYUS, 2023-04-17:
Hi @MartyG-RealSense  

[image]

The error occurs in cs-tutorial-capture.
error message : Method or operation not implemented.

The camera is also connected via USB. What's the problem?

MartyG-RealSense, 2023-04-17:
I will highlight your problem with cs-tutorial-capture to my Intel RealSense colleagues.  Thanks very much for your patience.

hsuys, 2023-04-18:
The cs-tutorial-capture example seems to work as expected on my system (D435, SDK2.53.1, VC2022).

closed 2023-05-01 by MartyG-RealSense