realsenseai/librealsense #11681: Why I cannot have stereo images and IMU measurement in ROS

Turing47, 2023-04-14:
I bought a D455 two years ago, and I used it several times in 2021. Recently I happen to need it.

I installed Realsense Viewer on Ubuntu 20 this week.
But I do not know how to get stereo images both Realsense Viewer and ROS . 
In 2021 I could get stereo images in both Realsense Viewer and ROS. Why I cannot get them now?

Also I can not find topics related to IMU in ROS. The topic list is shown as follow:
`/camera/color/camera_info
/camera/color/image_raw
/camera/color/image_raw/compressed
/camera/color/image_raw/compressed/parameter_descriptions
/camera/color/image_raw/compressed/parameter_updates
/camera/color/image_raw/compressedDepth
/camera/color/image_raw/compressedDepth/parameter_descriptions
/camera/color/image_raw/compressedDepth/parameter_updates
/camera/color/image_raw/theora
/camera/color/image_raw/theora/parameter_descriptions
/camera/color/image_raw/theora/parameter_updates
/camera/color/metadata
/camera/depth/camera_info
/camera/depth/image_rect_raw
/camera/depth/image_rect_raw/compressed
/camera/depth/image_rect_raw/compressed/parameter_descriptions
/camera/depth/image_rect_raw/compressed/parameter_updates
/camera/depth/image_rect_raw/compressedDepth
/camera/depth/image_rect_raw/compressedDepth/parameter_descriptions
/camera/depth/image_rect_raw/compressedDepth/parameter_updates
/camera/depth/image_rect_raw/theora
/camera/depth/image_rect_raw/theora/parameter_descriptions
/camera/depth/image_rect_raw/theora/parameter_updates
/camera/depth/metadata
/camera/extrinsics/depth_to_color
/camera/motion_module/parameter_descriptions
/camera/motion_module/parameter_updates
/camera/realsense2_camera_manager/bond
/camera/rgb_camera/auto_exposure_roi/parameter_descriptions
/camera/rgb_camera/auto_exposure_roi/parameter_updates
/camera/rgb_camera/parameter_descriptions
/camera/rgb_camera/parameter_updates
/camera/stereo_module/auto_exposure_roi/parameter_descriptions
/camera/stereo_module/auto_exposure_roi/parameter_updates
/camera/stereo_module/parameter_descriptions
/camera/stereo_module/parameter_updates
/diagnostics
/rosout
/rosout_agg
/tf
/tf_static
`

[image]

Can I get the stereo images and IMU measurement as previously?

MartyG-RealSense, 2023-04-15:
Do you mean that you want to have the depth and IMU streams enabled in both the RealSense Viewer and the ROS wrapper at the same time?  If you do then it is recommendable to use one or the other (Viewer or wrapper) but not both simultaneously.

Please try adding the commands below to the end of your roslaunch instruction to enable the Accel and Gyro IMU topics, which are disabled by default in the RealSense ROS wrapper.

enable_gyro:=true enable_accel:=true unite_imu_method:=linear_interpolation

The unite_imu_method command combines the two individual Accel and Gyro topics into a single /imu topic

Turing47, 2023-04-15:
Hi, @MartyG-RealSense
Thanks for your help. Now I know how to get stereo images and IMU data in ROS.

Here are two problems left:
I do not how to find stereo images in RealSense VIewer. Can you help me with that?
[image]

In ROS, I can find stereo images from topic "/camera/infra1/image_rect_raw" and "/camera/infra2/image_rect_raw". But why are there so many white points? How can I remove them?

[image]

Thank you

MartyG-RealSense, 2023-04-16:
In the RealSense Viewer, under the Stereo Module category there should be three streams listed on a USB 3 connection if you expand the category's options open by left-clicking on the arrow beside it: Depth, Infrared and Infrared 2.  

Infrared and Infrared 2 are the stereo left and right images.

[image]

The white dots are cast from the camera's IR Emitter component and help the camera to analyze objects and surfaces for depth detail by acting as a 'texture source'.  Deactivating the emitter can reduce the quality of the depth image, though having a strongly illuminated scene can compensate for this by enabling the camera to use the scene's ambient light for depth analysis instead of the dots.

You can deactivate the dots completely by changing the **Emitter Enabled** drop-down menu (shown in the image above) from its default setting of 'Laser' to 'Off'.

Alternatively, you can reduce the visibility of the dots but not remove them completely from the infrared image by reducing the value of the **Laser Power** option under Stereo Module > Controls.

Turing47, 2023-04-16:
Thanks for your detailed explanation. Your instruction is really helpful. Now I know how to solve the problem.  @MartyG-RealSense

MartyG-RealSense, 2023-04-16:
You are very welcome.  I'm pleased that I could help.  :)

closed 2023-04-28 by MartyG-RealSense