realsenseai/librealsense #11682: App freezes when starting to record on Android 13

Juris710, 2023-04-15:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 |
| Firmware Version                   | 05.14.00.00 |
| Operating System & Version |   Android 13 (Google Pixel 7 Pro)  |
| Kernel Version (Linux Only)    |  5.10.149  |
| Platform                                 |  Android  |
| SDK Version                            |  v2.53.1                          |
| Language                            |  Kotlin / Java |
| Segment			|  Smartphone |

### Issue Description

Hello.

I'm trying to record and save as .bag files on Google Pixel 7 Pro.
When I started recording, the app stopped responding.
I tried the latest commit of Realsense SDK on master branch and development branch, and the issue happened on both of them .
The firmware version of D435 is latest according to IntelRealsenseViewer.

I tried another phone running Android 11, and it worked.
So the issue may be related to Android OS version.

Thanks.

#### Steps to reproduce

1. Launch [recording example app](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android/examples/recording) or [RS Camera app](https://play.google.com/store/apps/details?id=com.intel.realsense.camera)
2. Connect realsense camera to phone
3. Start recording

#### Expected behavior

Recording starts.

#### Actual behavior

App freezes.

The following error was found in the log.
```
exception in UsbManager.openDevice
java.lang.SecurityException: User has not given 10347/com.intel.realsense.camera permission to access device /dev/bus/usb/003/002
	at android.os.Parcel.createExceptionOrNull(Parcel.java:3011)
	at android.os.Parcel.createException(Parcel.java:2995)
	at android.os.Parcel.readException(Parcel.java:2978)
	at android.os.Parcel.readException(Parcel.java:2920)
	at android.hardware.usb.IUsbManager$Stub$Proxy.openDevice(IUsbManager.java:932)
	at android.hardware.usb.UsbManager.openDevice(UsbManager.java:740)
	at com.intel.realsense.librealsense.DeviceWatcher.addDevice(DeviceWatcher.java:115)
	at com.intel.realsense.librealsense.DeviceWatcher.invalidateDevices(DeviceWatcher.java:78)
	at com.intel.realsense.librealsense.DeviceWatcher.access$000(DeviceWatcher.java:16)
	at com.intel.realsense.librealsense.DeviceWatcher$1.onDeviceAttach(DeviceWatcher.java:44)
	at com.intel.realsense.librealsense.Enumerator.notifyOnAttach(Enumerator.java:102)
	at com.intel.realsense.librealsense.Enumerator.access$200(Enumerator.java:18)
	at com.intel.realsense.librealsense.Enumerator$MessagesHandler.handleMessage(Enumerator.java:144)
	at android.os.Handler.dispatchMessage(Handler.java:106)
	at android.os.Looper.loopOnce(Looper.java:201)
	at android.os.Looper.loop(Looper.java:288)
	at android.os.HandlerThread.run(HandlerThread.java:67)
Caused by: android.os.RemoteException: Remote stack trace:
	at com.android.server.usb.UsbUserPermissionManager.checkPermission(UsbUserPermissionManager.java:678)
	at com.android.server.usb.UsbHostManager.openDevice(UsbHostManager.java:564)
	at com.android.server.usb.UsbService.openDevice(UsbService.java:293)
	at android.hardware.usb.IUsbManager$Stub.onTransact(IUsbManager.java:458)
	at android.os.Binder.execTransactInternal(Binder.java:1280)
```

MartyG-RealSense, 2023-04-15:
Hi @Juris710  The RealSense Android wrapper currently supports up to Android 11.  A RealSense user explored adding support for Android 12 at https://github.com/IntelRealSense/librealsense/pull/10965 but it is not currently part of the RealSense SDK.

As far as I am aware, Android 13 support has not been looked at yet.

You could try setting **targetSDKVersion** to a lower value in order to target an older Android version, as described at https://github.com/IntelRealSense/librealsense/issues/11190

Juris710, 2023-04-15:
Hi @MartyG-RealSense
I didn't know that Android 13 was not currently supported. I'm sorry.

Thank you for suggesting solutions.
I tried the fork #10965, but it didn't work.
Now I'm downgrading the targetSdkVersion as low as I can, but it does not seem to work, either.

Maybe I should wait until Android 13 is supported.
I'll use Android 11 phone instead.

Anyway, thank you for your help.
You can close this issue.

MartyG-RealSense, 2023-04-15:
You are very welcome.

closed 2023-04-15 by MartyG-RealSense

gloaguen, 2024-02-26:
@MartyG-RealSense what are the plans for Android 12, 13 and 14 support? Are the developers working is this? Can you give me an ETD?