realsenseai/librealsense #11683: Issue

Racer098, 2023-04-15:
| | |
|---|---|
|**librealsense**|2.53.1 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Unknown error (-529697949)!
```

MartyG-RealSense, 2023-04-15:
Hi @Racer098  Which RealSense camera model are you using, please?  Thanks!

Racer098, 2023-04-15:
Hi @MartyG-RealSense I am using Intel real sense depth camera D435i

MartyG-RealSense, 2023-04-15:
Thank you very much.  Does the error occur immediately after you plug the camera in, or when you enable the depth stream?

Racer098, 2023-04-15:
The error occurs when I try to turn on the RGB camera in the realsense viewer app after selecting source as platform

MartyG-RealSense, 2023-04-15:
To enable the RGB camera of your D435i, you should enable the **RGB** stream near the bottom of the Viewer's options side-panel.  The 'Plaform' option under Add Source is usually used for accessing non-RealSense cameras such as a laptop's built-in webcam.

Racer098, 2023-04-15:
Hey thanks for that response now I got a bit clarity, although I am unable to find or navigate to the option you are talking about could you please help me navigate to it on where can i find the viewers option side panel as I could see none in the real sense viewer app. Do I need to use a different application?

MartyG-RealSense, 2023-04-15:
If you have a D435i then the side-panel of the Viewer should look like the image below.  There is an 'RGB Camera' option with a red icon beside it.  Left-clicking on the red icon should enable the RGB stream.

[image]

If the D435i's RGB sensor is not detected by the camera then the RGB option would be missing and the camera name at the top of the side-panel would be **D430i** instead of D435i.

Racer098, 2023-04-15:
Hi, the camera is detected, I did what you said but it says no frames received.

MartyG-RealSense, 2023-04-15:
Does it say 'No Frames Received' if you reboot the computer, open the Viewer and then enable **only** RGB Camera?  

There is a known issue with the Viewer where enabling three streams (Depth, RGB and Motion Module) can cause the RGB panel to just be a white image with No Frames Received overlaid on it.

Racer098, 2023-04-15:
Hey, I did what you said and its working now!! thank you so much for your help and support really appreciate it

MartyG-RealSense, 2023-04-16:
You are very welcome  :)  As it is working for you now, I will close this case.  Thanks very much for the update!

closed 2023-04-16 by MartyG-RealSense