realsenseai/librealsense #11684: Empty files generated in tensorflow example4

georgekasa, 2023-04-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

python, 
Linux ubuntu 22.0
Python 3.9.16
intel 435i

problem
https://github.com/nohayassin/librealsense/blob/tensorflow/wrappers/tensorflow/example4%20-%20train.py
it does create just empty files, the 2 files which I downloaded 
part1_1_4000.zip
part2_4001_8375.zip.
I unzip them and they have only pictures inside not folders just pictures.

the only line of code which I altered is:
root = r"/media/gkasap/ssd256gb/datasets/intel435i/part1_1_4000"
any it creates empty files of images 

I think ther problem is it creates the new files like images_path

root = r"/media/gkasap/ssd256gb/datasets/intel435i/part1_1_4000"
images_path = root + r"/images"
models_path = root  + r"/models"
logs_path = root  + r"/logs"
train_images = images_path + r"/train"
train_cropped_images_path = images_path + r"/train_cropped"
paths = [root, images_path, models_path, logs_path, train_images, train_cropped_images_path]
print("Creating folders for training process ..")
for path in paths:
    if not os.path.exists(path):
        print("Creating  ", path)
        os.makedirs(path)

noisy_images = [f for f in glob.glob(test_images + "**/res*" + IMAGE_EXTENSION, recursive=True)]
ir_images = [f for f in glob.glob(test_images + "**/left*" + IMAGE_EXTENSION, recursive=True)]

and it search for the pictures inside the new file which we are just creating and nothing more.

MartyG-RealSense, 2023-04-16:
The GitHub link that you quoted is a custom 'forked' version of the TensorFlow wrapper rather than the official wrapper.  The official one received an update in February 2021 that your fork version does not have, in which http web addresses were replaced with secure **https** addresses.  This was a widespread change that Intel implemented to librealsense in early 2021.  The old http web links did not work after that.

It may therefore be worth first trying to install the official updated wrapper to see whether that resolves your problem.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/tensorflow

georgekasa, 2023-04-16:
python version will not have problem if it creates empty folder, tensorflow cannot read anything the files are just empty bellow:
images_path = root + r"/images"
models_path = root + r"/models"
logs_path = root + r"/logs"
train_images = images_path + r"/train"
train_cropped_images_path = images_path + r"/train_cropped"
##############################################
yes I download them again
it does create just empty files, the 2 files which I downloaded
part1_1_4000.zip
part2_4001_8375.zip.
##############################################
your link if and if I select example4, is the same author and the same code  I checked it with diff in terminal

MartyG-RealSense, 2023-04-16:
I am currently downloading the Example 1 link to check its contents.

georgekasa, 2023-04-16:
[image]

just pictures as I said it no files

MartyG-RealSense, 2023-04-16:
Thanks very much for your patience.  I confirmed that the contents of the zip are only image files and are not categorized into folders.

georgekasa, 2023-04-16:
root = r"/media/gkasap/ssd256gb/datasets/intel435i/part1_1_4000"
I only aletered this line, because I must set the where is the path

MartyG-RealSense, 2023-04-16:
As far as I am aware, no RealSense users have previously tried Example 4 and so there are no references regarding how to solve problems that might occur with it, unfortunately.  

Would it be possible for you to try Example 5, which also involves denoising? 

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/tensorflow#part-5---applying-trained-network-to-real-data

georgekasa, 2023-04-16:
error again.
python example5-denoised.py DEPTH_Keras_Unet.model
ValueError: Input 0 of layer "model" is incompatible with the layer: expected shape=(None, 128, 128, 2), found shape=(None, 480, 480, 2)

I presume your model is not trained with 128x128 shape but with 480x 480

MartyG-RealSense, 2023-04-16:
The creator of the RealSense TensorFlow wrapper - @nohayassin - is the best source of knowledge about this wrapper so I have tagged them into this discussion in case they can offer input about your Example 4 problem.

georgekasa, 2023-04-16:
thank you very much!!!

georgekasa, 2023-04-20:
also for some reason it has 2 classes 
channels = 2
img_width, img_height = 128, 128

unet_epochs = 1
conv10 = Conv2D(channels, 1, activation='sigmoid')(conv9)

model = Model(inputs=inputs, outputs=conv10)
model.compile(optimizer=Adam(lr=1e-4), loss='binary_crossentropy', metrics=['accuracy'])
and it uses sigmoid and binary_crossentropy should use 
 activation='softmax'
 and categorical_crossentropy 

Used for multi-class predictions. The sum of all outputs generated by softmax is 1.
https://www.tensorflow.org/api_docs/python/tf/nn/softmax
from chat gpt
Using binary_crossentropy as the loss function in this case might not be the most appropriate choice given that the number of channels is 2. Binary_crossentropy is usually used for binary classification problems, where you have only two classes.

MartyG-RealSense, 2023-04-26:
@nohayassin  As the creator of the RealSense TensorFlow wrapper, would you be able to offer @georgekasa advice about a problem they have with Example 4 please?  Thanks!

MartyG-RealSense, 2023-05-02:
Hi @georgekasa  I am sorry that you have not received a response to the request for help from the wrapper's creator with Example 4.  I do not know the answer myself, unfortunately.

MartyG-RealSense, 2023-05-12:
Hi @georgekasa  In response to your request for help, my Intel RealSense colleagues are testing the TensorFlow wrapper examples and discussing them.  Thanks very much for your patience!

MartyG-RealSense, 2023-05-24:
Thanks very much for your continued patience.  My Intel RealSense colleagues will continue to look at the Tensorflow wrapper after the next release of the RealSense SDK.  There is not a release date currently available for the next SDK, so there is unfortunately no further information that I can provide about whether the TensorFlow examples will receive an update.  I sincerely apologize.

georgekasa, 2023-05-24:
okay okay thanks for your update

MartyG-RealSense, 2023-06-08:
Hi @georgekasa  Intel have decided to remove the TensorFlow wrapper and not fix Examples 4 and 5, unfortunately.  Thanks so much again for the patience you have had whilst this case was being discussed and decided upon.

closed 2023-06-14 by MartyG-RealSense