realsenseai/librealsense #11685: Unable to install pyrealsense2 in jetson AGX Orin (arm64 architecture)

Bhanu5302, 2023-04-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I'm unable to install pyrealsense2 module in AGX orin (arm64) , can you give steps for how to install librealsense & pyrealsense2 on AGX orin (arm64) board

MartyG-RealSense, 2023-04-17:
Hi @Bhanu5302  A guide at https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049 for installing librealsense and pyrealsense2 on Jetson boards may be helpful to you.

MartyG-RealSense, 2023-04-23:
Hi @Bhanu5302  Do you require further assistance with this case, please?  Thanks!

Bhanu5302, 2023-04-26:
Thanks for the information

closed 2023-04-26 by Bhanu5302