realsenseai/librealsense #11686: rs-convert.exe

666tua, 2023-04-17:
I'm currently using Intelrealsense D455 for keypoint detection.I detect the two-dimensional coordinates on the color images.I had some problems getting depth information from two-dimensional coordinates:1. Alignment problem between color images and depth images.
    1. Alignment problem between color chart and depth chart. I recorded a 5 minute 5 second video using realsenseviewer. Then 2238 color maps and 2239 depth maps were obtained using rs-convert.exe.I obtained the two-dimensional coordinates of 2238 frames through the key point detection of the color graph.When I mapped 2,238 two-dimensional coordinates to the depth chart, I found that they were misaligned.
[image]
[image]
    2.The images produced by parsing bag files by rs-convert.exe and python are badly inconsistent.Since the image produced by rs-convert.exe is unaligned, I used the official python program to parse the bag file and  align the color and depth maps.But python parses 4460 color images and 4460 depth images.I tried to map my 2238 2D coordinates to the corresponding alignment depth map, but I couldn't!

MartyG-RealSense, 2023-04-17:
With method 1 (extracting the depth and color images and then mapping the coordinates together), the mis-alignment may be due to the different physical positions on the front of the camera of the depth and color sensors not being taken into account (with the color sensor being offset horizontally from the depth sensor).  The SDK's depth-color alignment would usually automatically make adjustments for that difference when mapping color and depth coordinates together by using **extrinsics** (the translation and rotation between two sensors).

Perhaps instead of recording a bag-file video of the individual color and depth streams in the 2D mode of the Viewer, you could map depth and color together in the Viewer's **3D** mode.  Instead of recording a bag, you could use a video screen-capture tool to record an .avi or .mp4 video and then extract the images from that video file so that you have a set of perfectly matched images.

666tua, 2023-04-21:
> With method 1 (extracting the depth and color images and then mapping the coordinates together), the mis-alignment may be due to the different physical positions on the front of the camera of the depth and color sensors not being taken into account (with the color sensor being offset horizontally from the depth sensor). The SDK's depth-color alignment would usually automatically make adjustments for that difference when mapping color and depth coordinates together by using **extrinsics** (the translation and rotation between two sensors).
> 
> Perhaps instead of recording a bag-file video of the individual color and depth streams in the 2D mode of the Viewer, you could map depth and color together in the Viewer's **3D** pointcloud mode. Instead of recording a bag, you could use a video screen-capture tool (such as Camtasia, Fraps or XSplit) to record an .avi or .mp4 video of the area of the Viewer's window that contains the pointcloud - or press the F8 keyboard key for a fullscreen view - and then extract the images from that video file so that you have a set of perfectly matched images.

Thanks for your quick reply, I have solved the problem.But when I get three dimensional coordinates, I always get zero. How do I solve for zero. There are no relevant papers to solve this problem.

MartyG-RealSense, 2023-04-21:
What method are you using to get the three-dimensional coordinates, please?

666tua, 2023-04-21:
> What method are you using to get the three-dimensional coordinates, please?

I tracked the mouse's two-dimensional key points on the color images.Then align the color images with the depth images.Finally, the get_distance () function provided by IntelRealsense SDK2.0 is used to obtain the distance information.

MartyG-RealSense, 2023-04-21:
You will not get XYZ coordinates without generating a 3D pointcloud, but you can retrieve the real world Z-distance in meters of a specific two-dimensional XY coordinate using get_distance().  A Python example of this can be found at https://github.com/IntelRealSense/librealsense/issues/9939

closed 2023-05-02 by MartyG-RealSense