realsenseai/librealsense #11687: Error running ORB_SLAM3 with D435i camera: invalid_value_error

saood-ahmad, 2023-04-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435I} | 
| Firmware Version                   | (05.14.00.00) | 
| Operating System & Version |   { Linux (Ubuntu 18) /   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC.  |
| SDK Version                            |  { legacy / 2.53.1 }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<I am trying to run ORB_SLAM3 with my Intel D435i camera using the command 
./Examples/Monocular/mono_realsense_D435i Vocabulary/ORBvoc.txt ./Examples/Monocular/RealSense_D435i.yaml...
However, I am encountering the following error message:
terminate called after throwing an instance of 'rs2::invalid_value_error'
  what():  hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
Aborted (core dumped)
Sure, based on the information you provided, here's an example of an issue message that you could post on the ORB_SLAM3 Github repository:

Title: Error running ORB_SLAM3 with D435i camera: invalid_value_error

Description:
I am trying to run ORB_SLAM3 with my Intel D435i camera using the command ./Examples/Monocular/mono_realsense_D435i Vocabulary/ORBvoc.txt ./Examples/Monocular/RealSense_D435i.yaml. However, I am encountering the following error message:

arduino
Copy code
terminate called after throwing an instance of 'rs2::invalid_value_error'
  what():  hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
Aborted (core dumped)
This error message occurs after installing the RealSense Viewer 2.5.3. Before installing the viewer, ORB_SLAM3 was working fine with my D435i camera.>

MartyG-RealSense, 2023-04-17:
Hi @saood-ahmad  Can you confirm please that your camera has firmware 5.14.0.0 installed, as this error might occur if you had SDK 2.53.1 installed but the camera had a firmware older than 5.14.0.0.

It may be worth considering downgrading to SDK 2.51.1 and firmware 5.13.0.50.

saood-ahmad, 2023-04-17:
It works ! Alhamdulillah 
thanks for your help

MartyG-RealSense, 2023-04-18:
It's great to hear that it worked for you.  Thanks very much for the update!

saood-ahmad, 2023-05-05:
Title: Auto Exposure not working with RealSense D435i camera in RGB-D mode

Description:

I am using ORB_SLAM3 with the RealSense D435i camera in RGB-D mode. When I run the command ./Examples/RGB-D/rgbd_realsense_D435i Vocabulary/ORBvoc.txt ./Examples/RGB-D/RealSense_D435i.yaml, the camera does not perform auto exposure.

I have tested the camera with other modes such as monocular and stereo, and auto exposure works fine. However, in RGB-D mode, auto exposure does not work.

I would appreciate it if someone could help me with this issue or provide any insights into why auto exposure is not working with the RealSense D435i camera in RGB-D mode.

Thank you.

MartyG-RealSense, 2023-05-06:
Circumstances in which auto-exposure could be deactivated by the contents of a .yaml configuration file are if (a) auto-exposure is set to false, or (b) a manual exposure value is set, which would cause auto-exposure to automatically disable.

MartyG-RealSense, 2023-05-12:
Hi @saood-ahmad  Do you require further assistance with this case, please?  Thanks!

saood-ahmad, 2023-05-12:
I have made modifications to the RealSense_D435i.yaml configuration file in an attempt to enable auto-exposure. However, I have not observed any effect on the camera's exposure behavior.

After investigating further, I checked the supported options for the camera, and it appears that auto-exposure is indeed available. I came across an issue mentioning that auto-exposure could be deactivated by the contents of the YAML configuration file if auto-exposure is set to false or a manual exposure value is set.

I have uploaded my RealSense_D435i.yaml file [https://drive.google.com/file/d/1prjbP7D-cmoKWUVwtZrf67JmoUxlicSD/view?usp=sharing](url)  for your reference.

##### Screenshots:

To provide a clearer understanding of the issue, I have captured screenshots in different lighting conditions, including low light and sunlight. The screenshots showcase the camera's output and the observed exposure settings for each lighting condition. You can find the screenshots [.](https://drive.google.com/file/d/1aW5ipvz15v1MpXnxi93nbtblt3KyqUyv/view?usp=sharing, https://drive.google.com/file/d/1bbhWFzUwS7Ld7O21g2UYNCcS3rg6YbXz/view?usp=sharing, https://drive.google.com/file/d/1oJe0vAP05ZMcqEwagM2GvwFbrfkwpuWF/view?usp=sharing)

MartyG-RealSense, 2023-05-12:
The ORB-SLAM3 yaml file does not have any configuration settings that would affect manual exposure or auto-exposure.

Thanks very much for the images of low-light and sunlight.  They resemble what an image may look like if the infrared sensors are saturated with light, which can occur if a scene is too dark or too sunny.  If auto-exposure is enabled then a way to test for sensor saturation is to cover your hand over the front of the camera for several seconds to give the auto-exposure time to make adjustments and then remove your hand and see whether the image auto-corrects.

https://github.com/IntelRealSense/librealsense/issues/2875 shows the effect that saturation can have on the infrared image in low-light and sunny conditions.

closed 2023-05-31 by MartyG-RealSense