realsenseai/librealsense #11688: jetson nano pyrealsense2 import error

bmos291, 2023-04-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455} | 
| Firmware Version                   | 05.14.00 | 
| Operating System & Version |   ubuntu 18.04  | 
| Kernel Version (Linux Only)    | 4.9.299                                         | 
| Platform                                 | Nvidia Jetson nano |
| SDK Version                            | 2.53.1                          | 
| Language                            | python 3.6                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Hello, I am using jetson nano. I used python3.7 before, but I changed the version to python3.6 to install pytorch.
Then, when I ran the code with python3.6, I got an error that I couldn't import pyrealsense2. In python3.7 pyrealsense2 is imported and works fine.
The pyrealsense2 folder exists in the /usr/local/lib/python3.7/ path, but of course it does not exist in python3.6. Is there a way to move pyrealsense2 to python3.6?

MartyG-RealSense, 2023-04-17:
Hi @bmos291  If you initially installed the pyrealsense2 wrapper for Python 3.7 and then changed your Python version to 3.6 without reinstalling the wrapper then you should install the wrapper again so that its files match with Python 3.6.

bmos291, 2023-04-17:
Should I delete the wrapper installed in Python 3.7? Or do I just install it as Python 3.6?
I tried to install python 3.6 again after following this instruction, but it failed with the following error.
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python

~/librealsense/build$ cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/local/lib/python3.6

-- -- pybind11 v2.6.2
-- -- Found PythonInterp: /usr/local/lib/python3.6 (found version "1.4")
CMake Error at build/third-party/pybind11/tools/FindPythonLibsNew.cmake:133 (message):
Python config failure:

I thought the full path of python executable (Python 3.6) was /usr/local/lib/python 3.6. How can I check full path? Is that /usr/bin/?

MartyG-RealSense, 2023-04-17:
You can check the path to your Python 3 installation with the command **which python3**

closed 2023-04-24 by bmos291