realsenseai/librealsense #11691: Setting up automatically my L515 without using Intel Realsense Viewer

CheCamCheCam, 2023-04-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | (01.05.08.01) | 
| Operating System & Version |   {Win (10)}  | 
| Platform                                 | PC  |
| SDK Version                            |  {  2 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {others }                          | 

### Issue Description

Hi everyone, 
I'm using my L515 camera in a C++ code, right now in order to set it up I open Intel Realsense Viewer and set it there on "short range", but I would like to automatically have it set this way through my code, I looked up at the librealsenseSDK library but I really don't understand how to use JSON file with it. I assumed that saving the config into a JSON file (only once on Intel Realsense Viewer) and then using a certain method would work, yet after reading and trying for a couple of hours I don't know how to set up a rs2::device according to my camera and how to import the informations from my JSON file to my code.
Could anyone provide any help please, 
best regards

MartyG-RealSense, 2023-04-17:
Hi @CheCamCheCam  The Visual Preset can be defined in C++ code instead of a preset.  I hope that https://github.com/IntelRealSense/librealsense/issues/9071#issuecomment-878552273 will be a helpful reference for implementing the L515 Short Range preset.

CheCamCheCam, 2023-04-18:
Hi, thank you for the link, I think it works as I managed to implement the lines in a way that doesn't cause errors and run the code but I wonder if there is a way to see if the modifications were made ? In any case thanks a lot for the help !

CheCamCheCam, 2023-04-19:
Thank you very much !