realsenseai/librealsense #11693: invalid write with customRW for underwater calibration on D405

RRHall, 2023-04-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D405                                                       | 
| Firmware Version                   | 5.13.00.50                                               | 
| Operating System & Version |   Linux   Nvidia Jetpack 5.0.2                  | 
| Kernel Version (Linux Only)    |  L4T-35.1.0 (5.10.104-tegra)                   | 
| Platform                                 | Nvidia Jetson Xavier NX 16Gb  |
| SDK Version                           |     Unsure - Installed using sudo apt-get install librscalibrationtool | 
| Language                               |  bash                 | 
| Segment			|  Robot                     | 

Hi Realsense

I'm trying to calibrate my D405 for an underwater environment. 
My process for doing so was the following 

1. Capture checkerboard images in water (tank) from unrectified raw streams from realsense
2. Extract subset of images
3. Put subset of images through MATLAB stereo calibration tool to get stereocamera parameters
4. Generate xml file
5. Rewrite stereo parameters back to realsense camera using customRW and xml file.

It has all worked successfully so far, except step 5. The calibration images were captured at resolution = 640x480.
MATLAB's calibration was performed at this resolution. I was unsure of what to set the RGB sensor so I calculated an average for focal length between left and right. 

What do I need to change to get this to work? 
~~~ 
Intel.Realsense.customRW -w -f calibration.xml
CustomRW for Intel RealSense D400, Version: 2.12.0.0

  Device PID: 0B5B
  Device name: Intel RealSense D405
  Serial number: 123622270320
  Firmware version: 05.13.00.50
  USB Type: 3.2

Error (9902): image resolution in calibration table on device is invalid.
Please re-calibrate the device with latest OEM Calibration Tool and try again.
check CustomRW tool.
~~~

~~~
calibration.xml
<?xml version="1.0"?>
<Config>
    <param name="ResolutionLeftRight">
        <value>640</value>
        <value>480</value>
    </param>
    <param name="FocalLengthLeft">
        <value>431.389</value>
        <value>515.081</value>
    </param>
    <param name="PrincipalPointLeft">
        <value>319.458</value>
        <value>238.477</value>
    </param>
    <param name="DistortionLeft">
        <value>0.3123</value>
        <value>0.2319</value>
        <value>-0.0028</value>
        <value>0.0014</value>
        <value>0.4485</value>
    </param>
    <param name="FocalLengthRight">
        <value>431.216</value>
        <value>514.964</value>
    </param>
    <param name="PrincipalPointRight">
        <value>319.923</value>
        <value>236.128</value>
    </param>
    <param name="DistortionRight">
        <value>0.3069</value>
        <value>0.2990</value>
        <value>-0.0048</value>
        <value>0.0035</value>
        <value>0.2628</value>
    </param>
    <param name="RotationLeftRight">
        <value>1.0</value>
        <value>-0.0025</value>
        <value>-0.0038</value>
        <value>0.0025</value>
        <value>1.0</value>
        <value>0.004</value>
        <value>0.0038</value>
        <value>-0.004</value>
        <value>1.0</value>
    </param>
    <param name="TranslationLeftRight">
        <value>-17.248</value>
        <value>-0.000574</value>
        <value>0.1378</value>
    </param>
    <param name="HasRGB">
        <value>1</value>
    </param>
    <param name="ResolutionRGB">
        <value>640</value>
        <value>480</value>
    </param>
    <param name="FocalLengthRGB">
        <value>431.302</value>
        <value>515.022</value>
    </param>
    <param name="PrincipalPointRGB">
        <value>319.690</value>
        <value>237.303</value>
    </param>
    <param name="DistortionRGB">
        <value>0</value>
        <value>0</value>
        <value>0</value>
        <value>0</value>
        <value>0</value>
    </param>
    <param name="RotationLeftRGB">
        <value>0.999995</value>
        <value>-0.000964127</value>
        <value>0.00308445</value>
        <value>0.000979274</value>
        <value>0.999987</value>
        <value>-0.00491298</value>
        <value>-0.00307968</value>
        <value>0.00491597</value>
        <value>0.999983</value>
    </param>
    <param name="TranslationLeftRGB">
        <value>14.7957</value>
        <value>0.124071</value>
        <value>-0.505658</value>
    </param>
~~~

MartyG-RealSense, 2023-04-18:
Hi @RRHall  The calibration resolution is **1920x1080**, as shown in the example calibration xml on page 61 of the Dynamic Calibration user guide.

https://www.intel.com/content/www/us/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html

The RGB section of the page is quoted below.

[image]

**********

The D405 does not have a separate RGB sensor however.  Instead, its RGB is provided by the depth sensor.  The absence of an RGB sensor helps to enable the D405's compact size.

RRHall, 2023-04-18:
Hi @MartyG-RealSense 

I included the rgb sensor in the xml file because the Intel.realsense.CustomRW tool gave it to me when I read the calibration parameters into a backup file. I copied the backup and adjusted where necessary but tried to leave it as similar as possible. 

I have attempted to write the calibration parameters using the 1920x1080 resolution. 
I took the calibration params (solved for 640x480) and scaled them to 1920x1080. 
I then typed those values into the xml file and loaded it using the tool.
It successfully writes the calibration parameters. HOWEVER when I attempt to get frames using the SDK in a C++ program, I receive a 15 second timeout runtime error and no frames are retrieved. 

The most recent bit of progress I made was writing this xml file with the rgb sensor portion commented out. In the image is the realsense viewer displaying a point cloud of a flat wall at my desk. You can see it inverts the typical distortion found when generating point clouds underwater. **This is good! I was testing this in air so this is expected behaviour!**. However, when I tried to get frames inside the same c++ program, I could only see rgb images, depth images were blank. 
I'm unsure of where the current problem lies.
From this point forward I've got a few different avenues to take:
1. Check whether the missing depth frame is a type mismatch or scaling issue inside the program.
2. Revert calibration to factory and measure planar objects at different distances underwater to generate an inverse distortion profile I can apply to each image. 
3. Keep playing with the calibration xml files and see if something changes

Obviously I would really appreciate any input from Intel staff or other users who may have run into this problem. Please reply with any information. 

[image]

~~~
<?xml version="1.0"?>
<Config>  
    <param name="ResolutionLeftRight">  
        <value>640</value>  
        <value>480</value>  
    </param>  
    <param name="FocalLengthLeft">  
        <value>431.389</value>  
        <value>515.081</value>  
    </param>  
    <param name="PrincipalPointLeft">  
        <value>319.458</value>  
        <value>238.477</value>  
    </param>  
    <param name="DistortionLeft">  
        <value>0.3123</value>
        <value>0.2319</value>
        <value>-0.0028</value>
        <value>0.0014</value>
        <value>0.4485</value>
    </param>
    <param name="FocalLengthRight">
        <value>431.216</value>
        <value>514.964</value>
    </param>
    <param name="PrincipalPointRight">
        <value>319.923</value>
        <value>236.128</value>
    </param>
    <param name="DistortionRight">
        <value>0.3069</value>
        <value>0.2990</value>
        <value>-0.0048</value>
        <value>0.0035</value>
        <value>0.2628</value>
    </param>
    <param name="RotationLeftRight">
        <value>1.0</value>
        <value>-0.0025</value>
        <value>-0.0038</value>
        <value>0.0025</value>
        <value>1.0</value>
        <value>0.004</value>
        <value>0.0038</value>
        <value>-0.004</value>
        <value>1.0</value>
    </param>
    <param name="TranslationLeftRight">
        <value>-17.248</value>
        <value>-0.000574</value>
        <value>0.1378</value>
    </param>
    <!-- <param name="HasRGB">
        <value>1</value>
    </param>
    <param name="ResolutionRGB">
        <value>640</value>
        <value>480</value>
    </param>
    <param name="FocalLengthRGB">
        <value>431.302</value>
        <value>515.022</value>
    </param>
    <param name="PrincipalPointRGB">
        <value>319.690</value>
        <value>237.303</value>
    </param>
    <param name="DistortionRGB">
        <value>0</value>
        <value>0</value>
        <value>0</value>
        <value>0</value>
        <value>0</value>
    </param>
    <param name="RotationLeftRGB">
        <value>0.999995</value>
        <value>-0.000964127</value>
        <value>0.00308445</value>
        <value>0.000979274</value>
        <value>0.999987</value>
        <value>-0.00491298</value>
        <value>-0.00307968</value>
        <value>0.00491597</value>
        <value>0.999983</value>
    </param>
    <param name="TranslationLeftRGB">
        <value>14.7957</value>
        <value>0.124071</value>
        <value>-0.505658</value>
    </param> -->
</Config>
~~~

MartyG-RealSense, 2023-04-18:
The best available technical discussion about underwater calibration is at https://github.com/IntelRealSense/realsense-ros/issues/1723 where a senior Intel RealSense team member provides guidance.

The discussion is so long and detailed that a lot of the information is hidden under a cut.  So if you have not already read that case, I recommend visiting https://github.com/IntelRealSense/realsense-ros/issues/1723#issuecomment-802741234 and left-clicking on the **Load More** link under that linked-to comment, then returning to the start of the discussion.

The RealSense user in that case also reported the depth curvature effect at https://github.com/IntelRealSense/realsense-ros/issues/1723#issuecomment-789702949

RRHall, 2023-04-24:
Hi Marty

Thanks for linking that discussion. That does provide some really useful information. However, it seems the solution to this discussion was to buy the OEM calibration tool. Unfortunately, I do not have time or resources to do that. 
If I could reiterate, the problem is with uploading the calibration parameters to the camera.
Using MATLAB's stereo calibration toolbox, I calculated the intrinsic matrix and distortion parameters for each camera. 
I arranged these values into an xml file that meets the syntax for loading into the D405. 

However, each time I load the calibration into the camera it complains and returns an error. Can you help me understand how and why this occurs as well as if there is a way to fix it?
[image]

MartyG-RealSense, 2023-04-24:
Hi @RRHall  You may be able to use a print-out of a target image using the free version of the Dynamic Calibration tool and seal the image inside a transparent, non-reflective waterproof protector if the target has to be placed underwater.  The print target can be downloaded as a PDF from the link below and printed off.

https://dev.intelrealsense.com/docs/dynamic-calibration-print-target

Calibrating underwater and then writing the calibration to the camera is likely to be much easier than working out calibration values in MATLAB and then editing an xml file and attempting to write it to the camera.  Even if Dynamic Calibration does not calibrate the RGB, it is unlikely to have a significant negative effect as the RGB on D405 comes from the same sensor as depth (as there is not a separate RGB sensor) and so depth and RGB should already be in alignment by default.

MartyG-RealSense, 2023-04-30:
Hi @RRHall  Do you require further assistance with this case, please?  Thanks!

RRHall, 2023-05-01:
Hi @MartyG-RealSense 

I do require assistance with this case, but it is no longer urgent. When I upload my calibration to the camera, I receive the following error message. However, the custom calibration is saved onto the camera hardware and all images retrieved from it are rectified according to the parameters in the custom calibration. 

Given the calibration is successful and functions well underwater, I'm going to close the issue. Nevertheless, it would be good to remove this error message on upload. 
[image]

closed 2023-05-01 by RRHall

MartyG-RealSense, 2023-05-01:
It's great to hear that the custom calibration is successfully saved.  Thanks very much for the update!

RRHall, 2023-05-03:
Hi @fredwrnr 

I used the xml file I posted in this response.
In the MATLAB calibration my dataset was saved at 640,480 resolution. Which is why I put 640,480 into the xml file. 
If you put this xml example into a text editor (e.g. vscode) you'll notice I have commented out the RGB sensor calibration. 
I'm unsure if any of this is correct practice. It just happened to work for me.

> Hi @MartyG-RealSense
> 
> I included the rgb sensor in the xml file because the Intel.realsense.CustomRW tool gave it to me when I read the calibration parameters into a backup file. I copied the backup and adjusted where necessary but tried to leave it as similar as possible.
> 
> I have attempted to write the calibration parameters using the 1920x1080 resolution. I took the calibration params (solved for 640x480) and scaled them to 1920x1080. I then typed those values into the xml file and loaded it using the tool. It successfully writes the calibration parameters. HOWEVER when I attempt to get frames using the SDK in a C++ program, I receive a 15 second timeout runtime error and no frames are retrieved.
> 
> The most recent bit of progress I made was writing this xml file with the rgb sensor portion commented out. In the image is a top down view of the realsense viewer displaying a point cloud of a flat wall at my desk. You can see it inverts the typical distortion found when generating point clouds underwater. **This is good! I was testing this in air so this is expected behaviour!**. However, when I tried to get frames inside the same c++ program, I could only see rgb images, depth images were blank. I'm unsure of where the current problem lies. From this point forward I've got a few different avenues to take:
> 
> 1. Check whether the missing depth frame is a type mismatch or scaling issue inside the program.
> 2. Revert calibration to factory and measure planar objects at different distances underwater to generate an inverse distortion profile I can apply to each image.
> 3. Keep playing with the calibration xml files and see if something changes
> 
> Obviously I would really appreciate any input from Intel staff or other users who may have run into this problem. Please reply with any information.
> 
> [image]
> 
> ```
> <?xml version="1.0"?>
> <Config>  
>     <param name="ResolutionLeftRight">  
>         <value>640</value>  
>         <value>480</value>  
>     </param>  
>     <param name="FocalLengthLeft">  
>         <value>431.389</value>  
>         <value>515.081</value>  
>     </param>  
>     <param name="PrincipalPointLeft">  
>         <value>319.458</value>  
>         <value>238.477</value>  
>     </param>  
>     <param name="DistortionLeft">  
>         <value>0.3123</value>
>         <value>0.2319</value>
>         <value>-0.0028</value>
>         <value>0.0014</value>
>         <value>0.4485</value>
>     </param>
>     <param name="FocalLengthRight">
>         <value>431.216</value>
>         <value>514.964</value>
>     </param>
>     <param name="PrincipalPointRight">
>         <value>319.923</value>
>         <value>236.128</value>
>     </param>
>     <param name="DistortionRight">
>         <value>0.3069</value>
>         <value>0.2990</value>
>         <value>-0.0048</value>
>         <value>0.0035</value>
>         <value>0.2628</value>
>     </param>
>     <param name="RotationLeftRight">
>         <value>1.0</value>
>         <value>-0.0025</value>
>         <value>-0.0038</value>
>         <value>0.0025</value>
>         <value>1.0</value>
>         <value>0.004</value>
>         <value>0.0038</value>
>         <value>-0.004</value>
>         <value>1.0</value>
>     </param>
>     <param name="TranslationLeftRight">
>         <value>-17.248</value>
>         <value>-0.000574</value>
>         <value>0.1378</value>
>     </param>
>     
> </Config>
> ```

MartyG-RealSense, 2023-05-04:
Hi @fredwrnr  Are you using the D405 camera model like @RRHall is?  The D405 is not equipped with a separate RGB sensor component and instead obtains RGB from the depth sensor.

fredwrnr, 2023-05-04:
@MartyG-RealSense  yes I'm using the D405. 
In the xml-File it says:
<param name = "HasRGB">
        <value>1</value>
    </param>
thats why i was wondering. 
But if the camera obtains RGB from the IR-sensor, how is the RGB-image still distorted when the IR-image is not? 
For the rgb-image I'm streaming BGR8 and the IR-Image is Y8.

MartyG-RealSense, 2023-05-04:
The D405's RGB image is processed through an Image Signal Processor (ISP) hardware component inside the camera.

fredwrnr, 2023-05-04:
So what would be the way to calibrate the RGB-image?

MartyG-RealSense, 2023-05-04:
My colleagues confirmed that RGB is calibrated automatically when depth is calibrated as the RGB uses the same left IR sensor that depth originates from.

fredwrnr, 2023-05-04:
Okay, thanks for asking. I dont understand how its possible, that the RGB-Image is distorted when the IR-Image is not, as per the screenshot in my previous message. Do you have any idea how to solve that problem?

MartyG-RealSense, 2023-05-05:
On earlier RealSense models such as D415 and D435 / D435i, RGB had its 5 distortion coefficients all set to zero.  On later models including D455 and D405, the distortion coefficients for color are non-zero.  The image below showing the **coeffs** is from my own D405.

[image]

The distortion coefficients for infrared on D405 are still all-zero though.

[image]

fredwrnr, 2023-05-06:
When I list the intrinsics of the ir-sensor with customRW.exe they're not all-zero. 
But this still doesnt answer why when i write the distortion coeffs with customRW.exe, only the infrared-images are taking those new coeffs but the rgb-image doesnt, even though the values are updated when I list them afterwards. Do you have any information on why that is, or how I could maybe fix it?

MartyG-RealSense, 2023-05-06:
As RGB is calibrated automatically on D405 when depth is calibrated because the camera does not have a separate RGB sensor, it is possible that it does not respond to a non-realtime manual edit of the RGB values.  This is speculation on my part though as there is not information available about how the Dynamic Calibration tool handles RGB calibration on D405.