realsenseai/librealsense #11698: pyrealsense installs to python2 instead of python3

eladpar, 2023-04-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | 2.50.0 | 
| Operating System & Version |  Ubuntu 20  | 
| Kernel Version (Linux Only)    |  5.10.104-tegra                                        | 
| Platform                                 | NVIDIA jetson xavier NX |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2023-04-18:
Hi @eladpar  I recommend using the Jetson installation guide at https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049 which is further down the page that you linked to.

eladpar, 2023-04-18:
> Hi @eladpar I recommend using the Jetson installation guide at [#6964 (comment)](https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049) which is further down the page that you linked to.

Sorry my link was wrong I did use the one you sent

MartyG-RealSense, 2023-04-18:
**/pybackend2.cpython-38-aarch64-linux-gnu.so** is a filename that would be created when building the pyrealsense2 wrapper if Python 3.8 is installed on your computer.

Have you performed steps 7 and 8 - setting the PYTHONPATH and sourcing the bashrc file?

eladpar, 2023-04-18:
> **/pybackend2.cpython-38-aarch64-linux-gnu.so** is a filename that would be created when building the pyrealsense2 wrapper if Python 3.8 is installed on your computer.
> 
> Have you performed steps 7 and 8 - setting the PYTHONPATH and sourcing the bashrc file?

But i'm in step 6 ...
your'e saying it did detect im using python3.8 but still put the artifacts in the wrong folder

MartyG-RealSense, 2023-04-18:
In the guide there is a typing mistake where there should not be a space between '-' and 'DPYTHON' in - DPYTHON_EXECUTABLE=/usr/bin/python3.6

Did you close up the space between - and DPYTHON when using the CMake build instruction?  And did you modify the instruction to say python3.8 instead of python3.6 if 3.8 is the Python version that is installed on your computer?

You can confirm which Python 3 version you have with the command **which python3**

eladpar, 2023-04-18:
> In the guide there is a typing mistake where there should not be a space between '-' and 'DPYTHON' in:
>
> `- DPYTHON_EXECUTABLE=/usr/bin/python3.6`
>
> Did you close up the space between - and DPYTHON [...]

I changed it to be '-DPYTHON_EXECUTABLE=/usr/bin/python3' and I checked that my python3 executable is '/usr/bin/python3'

MartyG-RealSense, 2023-04-29:
Case closed due to no further comments received.

closed 2023-04-29 by MartyG-RealSense

IricEmage, 2023-05-25:
I have the same issue, it install into python2.7 instead of python3, I follow the link you send

MartyG-RealSense, 2023-05-25:
Hi @IricEmage  Please create a new issue for your problem by visiting the front page of this support forum at the link below and clicking the green **New Issue** button.  Thanks!

https://github.com/IntelRealSense/librealsense/issues/

IricEmage, 2023-05-25:
I got it. Thanks you