realsenseai/librealsense #11701: Images are saved as 320x240 even though in setting and calibration set to 640x480

monajalal, 2023-04-18:
I have an intel realsense 435D.
I am interested in simulataneusly capturing an image of both depth and RGB. I have this camera calibration in MORE --> Camera Calibration set for the Rectified Resolution of 640x480.

Then when I capture a photo (currently I am not able to take simultaneos photos of depth and RGB --appreciate if you could provide any pointers) for depth, I get this in the CSV even though in Intel RealSense Viewer v2.50.11 I see the following:

[image]

```
(base) mona@ard-gpu-01:~/test435/SPAM/640$ cat wq_Depth_Depth_metadata.csv
Frame Info: 
Type,Depth
Format,Z16
Frame Number,14661
Timestamp (ms),1681849042763.90
Resolution x,320
Resolution y,240
Bytes per pixel,2

Intrinsic:,
Fx,192.620239
Fy,192.620239
PPx,159.609146
PPy,118.386765
Distorsion,Brown Conrady

```
Essentially, could you please help with the followings:
1. force the taken depth image to be of size 640x480
2. simultaneously take photos of both depth and rgb by one click

[image]

[image]
So right now above the RGB and above the depth there is each a button for saving the image. When I click on Frame MetaData for depth, I see this:

[image]

How can I change the display size to 640x480?

* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

[image]

```
(base) mona@ard-gpu-01:~$ uname -a
Linux ard-gpu-01 5.19.0-38-generic #39~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Fri Mar 17 21:16:15 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
(base) mona@ard-gpu-01:~$ lsb_release -a
LSB Version:	core-11.1.0ubuntu4-noarch:security-11.1.0ubuntu4-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.1 LTS
Release:	22.04
Codename:	jammy

```
### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2023-04-19:
Hi @monajalal  The RealSense Viewer's information overlay shows the depth resolution as half of the size that was defined because a post-processing filter called the **Decimation Filter** is enabled by default.  The Decimation Filter reduces (or 'downsamples') the resolution by dividing it by a set value.  The default division value for the filter is '2', so 640x480 becomes 320x240.

You can display the full defined resolution by going to the **Stereo Module > Post Processing** section of the Viewer's options side-panel, expanding open the list of filters and left-clicking on the blue icon (On) beside 'Decimation Filter' to turn the icon to red (Off).  The resolution is then shown at its defined size in the overlay.

[image]

MartyG-RealSense, 2023-04-20:
You are very welcome, @monajalal - I'm pleased to hear that you achieved a solution in Python.  Thanks very much for the update!