realsenseai/librealsense #11702: librealsense on Ubuntu 22.04 -- error: building libusb:x64-linux failed with: BUILD_FAILED

monajalal, 2023-04-18:
```
(base) mona@ard-gpu-01:~$ git clone https://github.com/Microsoft/vcpkg.git
Cloning into 'vcpkg'...
remote: Enumerating objects: 189942, done.
remote: Counting objects: 100% (1354/1354), done.
remote: Compressing objects: 100% (442/442), done.
remote: Total 189942 (delta 928), reused 1306 (delta 912), pack-reused 188588
Receiving objects: 100% (189942/189942), 70.84 MiB | 5.05 MiB/s, done.
Resolving deltas: 100% (122259/122259), done.
(base) mona@ard-gpu-01:~$ cd vcpkg
(base) mona@ard-gpu-01:~/vcpkg$ ./bootstrap-vcpkg.sh
Downloading vcpkg-glibc...
Telemetry
---------
vcpkg collects usage data in order to help us improve your experience.
The data collected by Microsoft is anonymous.
You can opt-out of telemetry by re-running the bootstrap-vcpkg script with -disableMetrics,
passing --disable-metrics to vcpkg on the command line,
or by setting the VCPKG_DISABLE_METRICS environment variable.

Read more about vcpkg telemetry at docs/about/privacy.md
(base) mona@ard-gpu-01:~/vcpkg$ ./vcpkg integrate install
Applied user-wide integration for this vcpkg root.
CMake projects should use: "-DCMAKE_TOOLCHAIN_FILE=/home/mona/vcpkg/scripts/buildsystems/vcpkg.cmake"
(base) mona@ard-gpu-01:~/vcpkg$ ./vcpkg install realsense2
Computing installation plan...
A suitable version of cmake was not found (required v3.25.1) Downloading portable cmake 3.25.1...
Downloading cmake...
https://github.com/Kitware/CMake/releases/download/v3.25.1/cmake-3.25.1-linux-x86_64.tar.gz->/home/mona/vcpkg/downloads/cmake-3.25.1-linux-x86_64.tar.gz
Extracting cmake...
The following packages will be built and installed:
  * libusb[core]:x64-linux -> 1.0.26.11791
    realsense2[core]:x64-linux -> 2.53.1
  * vcpkg-cmake[core]:x64-linux -> 2022-12-22
  * vcpkg-cmake-config[core]:x64-linux -> 2022-02-06#1
Additional packages (*) will be modified to complete this operation.
Detecting compiler hash for triplet x64-linux...
Restored 0 package(s) from /home/mona/.cache/vcpkg/archives in 13.1 us. Use --debug to see more details.
Installing 1/4 libusb:x64-linux...
Building libusb[core]:x64-linux...
libusb currently requires the following tools and libraries from the system package manager:
    autoreconf
    libudev

These can be installed on Ubuntu systems via apt-get install autoreconf libudev-dev
-- Downloading https://github.com/libusb/libusb/archive/fcf0c710ef5911ae37fbbf1b39d48a89f6f14e8a.tar.gz -> libusb-libusb-fcf0c710ef5911ae37fbbf1b39d48a89f6f14e8a.tar.gz...
-- Extracting source /home/mona/vcpkg/downloads/libusb-libusb-fcf0c710ef5911ae37fbbf1b39d48a89f6f14e8a.tar.gz
-- Using source at /home/mona/vcpkg/buildtrees/libusb/src/89f6f14e8a-1dccf18d37.clean
-- Getting CMake variables for x64-linux-dbg
-- Getting CMake variables for x64-linux-rel
-- Generating configure for x64-linux
-- Finished generating configure for x64-linux
-- Configuring x64-linux-dbg
CMake Error at scripts/cmake/vcpkg_execute_required_process.cmake:112 (message):
    Command failed: /usr/bin/bash -c "V=1 ./../src/89f6f14e8a-1dccf18d37.clean/configure  \"--disable-silent-rules\" \"--verbose\" \"--disable-shared\" \"--enable-static\" \"--prefix=/home/mona/vcpkg/installed/x64-linux/debug\" \"--bindir=\\${prefix}/../tools/libusb/debug/bin\" \"--sbindir=\\${prefix}/../tools/libusb/debug/sbin\" \"--libdir=\\${prefix}/lib\" \"--includedir=\\${prefix}/../include\" \"--datarootdir=\\${prefix}/share/libusb\""
    Working Directory: /home/mona/vcpkg/buildtrees/libusb/x64-linux-dbg
    Error code: 1
    See logs for more information:
      /home/mona/vcpkg/buildtrees/libusb/config-x64-linux-dbg-config.log
      /home/mona/vcpkg/buildtrees/libusb/config-x64-linux-dbg-out.log
      /home/mona/vcpkg/buildtrees/libusb/config-x64-linux-dbg-err.log

Call Stack (most recent call first):
  scripts/cmake/vcpkg_configure_make.cmake:829 (vcpkg_execute_required_process)
  ports/libusb/portfile.cmake:58 (vcpkg_configure_make)
  scripts/ports.cmake:147 (include)

error: building libusb:x64-linux failed with: BUILD_FAILED
Please ensure you're using the latest port files with `git pull` and `vcpkg update`.
Then check for known issues at:
    https://github.com/microsoft/vcpkg/issues?q=is%3Aissue+is%3Aopen+in%3Atitle+libusb
You can submit a new issue at:
    https://github.com/microsoft/vcpkg/issues/new?title=[libusb]+Build+error&body=Copy+issue+body+from+%2Fhome%2Fmona%2Fvcpkg%2Finstalled%2Fvcpkg%2Fissue_body.md
```

[image]

* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

```
(base) mona@ard-gpu-01:~$ uname -a
Linux ard-gpu-01 5.19.0-38-generic #39~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Fri Mar 17 21:16:15 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
(base) mona@ard-gpu-01:~$ lsb_release -a
LSB Version:	core-11.1.0ubuntu4-noarch:security-11.1.0ubuntu4-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.1 LTS
Release:	22.04
Codename:	jammy

```

Camera: Intel RealSense D435

monajalal, 2023-04-18:
pyrealsense2 is at least installed using pip do I still need to deal with vcpkg?

https://github.com/IntelRealSense/librealsense/issues/3062#issuecomment-454002458

MartyG-RealSense, 2023-04-19:
Hi @monajalal  I strongly recommend not using vcpkg for RealSense installation if possible and installing from packages or source code instead.

If you have successfully installed the librealsense SDK and the pyrealsense2 wrapper has been successfully installed with **pip install pyrealsense2** then you should not need to deal with vcpkg.

monajalal, 2023-04-19:
Thank you @MartyG-RealSense I have no problem with lib. Currently my problem is I saw images/videos at 640x480 but they are saved differently. Is that a known bug?

closed 2023-04-19 by monajalal

MartyG-RealSense, 2023-04-19:
Do you mean that you save depth and color to a bag but they are saved as separate streams and not aligned?  If so then this is normal as RealSense SDK bag files do not save aligned data.  To perform alignment on bag-stored individual streams, the bag should be loaded into memory and then alignment performed on the two streams in real-time.