realsenseai/librealsense #11704: Intel realsense viewer errorcode

mitsuki0222, 2023-04-19:
When starting up an application, I get the following error message and the application does not start up. What should I do?
| | |

|---|---|
|**librealsense**|2.49.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
GL_INVALID_VALUE - C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\common\opengl3.cpp:569

```

MartyG-RealSense, 2023-04-19:
Hi @mitsuki0222  A **GL_INVALID_VALUE** suggests a problem related to OpenGL graphics support on your computer.  In past occurrences of this error that have been reported, a solution has been to download and launch an old version of the RealSense Viewer, then close it and launch the most recent Viewer.  This process is described at https://github.com/IntelRealSense/librealsense/issues/10544#issuecomment-1138569631

mitsuki0222, 2023-04-19:
Hi @MartyG-RealSense 
Thank you for replying.
When I pressed download old version it took me to the link below and I could not download it. What should I do?

https://camo.githubusercontent.com/81d3305a261a6bce96e73999c25186be03dd26200ef8b3532635b71f7996a930/68747470733a2f2f696d672e736869656c64732e696f2f6769746875622f646f776e6c6f6164732f496e74656c5265616c53656e73652f6c69627265616c73656e73652f76322e32312e302f746f74616c2e737667

MartyG-RealSense, 2023-04-19:
It looks as though you clicked on the 'Downloads' graphic at the top of the page.  This is not the download link.  You should scroll down the 2.21.0 page to the bottom and left-click on the link **Intel.RealSense.Viewer.exe**

Here is a direct download link for that file.

https://github.com/IntelRealSense/librealsense/releases/download/v2.21.0/Intel.RealSense.Viewer.exe

MartyG-RealSense, 2023-04-19:
You are very welcome  :)

closed 2023-04-25 by MartyG-RealSense