realsenseai/librealsense #11710: I ran into problems developing the L515

0DWADE0, 2023-04-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L500 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10)| 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I added the library files and header files needed for realsense development to the project I created, but the following error occurred and I can't fix it.

[image]
].

MartyG-RealSense, 2023-04-20:
Hi @0DWADE0  It looks like **LNK2001** errors with the GLFW3 graphics components.  Following the advice at https://github.com/IntelRealSense/librealsense/issues/3304#issuecomment-466197624 may therefore help to resolve the issue.

0DWADE0, 2023-04-20:
Hi@MartyG-RealSense I have read your suggestion, but I still can't solve my problem. I have added the additional dependencies and library directory. I downloaded glfw-3.3.8.bin.WIN32 from the official website.

MartyG-RealSense, 2023-04-20:
If you are using Visual Studio then instead of adding the linkages manually, it may be easier to add them using three **.props** property sheet files provided by the RealSense SDK.  Instructions for using the .props files can be found at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/4412175411859/comments/4412182682387

MartyG-RealSense, 2023-05-02:
Case closed due to no further comments received.

closed 2023-05-02 by MartyG-RealSense

0DWADE0, 2023-05-05:
Hi @MartyG-RealSense I solved the last problem, but I have a new one
[image]
Similarly, I also added opencv library, but this problem still occurs

MartyG-RealSense, 2023-05-05:
Another L515 user at https://github.com/IntelRealSense/librealsense/issues/7081#issuecomment-673676785 had this issue with debug_build_guard.