realsenseai/librealsense #11712: librscalibrationtool not available in ubuntu 22 (jammy) apt repository

jomiham, 2023-04-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | - | 
| Operating System & Version |   Linux (Ubuntu 22.04) | 
| Kernel Version (Linux Only)    |  5.15.0-69-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.53.1                        | 
| Language                            |  -                          | 
| Segment			|  -                          | 

### Issue Description
Trying to install the Dynamic Calibration Tool on Ubuntu 22.04 (jammy) by following the instructions here:
https://www.intel.com/content/www/us/en/download/645988/intel-realsense-d400-series-dynamic-calibration-tool.html

Ubuntu 22.04 is listed as supported, but there doesn't seem to be a `librscalibrationtool` package in the jammy repository:
```
> sudo apt-get install librscalibrationtool
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package librscalibrationtool
```

```
> curl https://librealsense.intel.com/Debian/apt-repo/dists/jammy/main/binary-amd64/Packages 2>&1 | grep Package | awk '{print $2}' | sort -u
librealsense2
librealsense2-dbg
librealsense2-dev
librealsense2-dkms
librealsense2-gl
librealsense2-gl-dbg
librealsense2-gl-dev
librealsense2-net
librealsense2-net-dbg
librealsense2-net-dev
librealsense2-udev-rules
librealsense2-utils
```

It is there on focal, bionic and xenial though.
```
> curl https://librealsense.intel.com/Debian/apt-repo/dists/focal/main/binary-amd64/Packages 2>&1 | grep Package | awk '{print $2}' | sort -u
librealsense2
librealsense2-dbg
librealsense2-dev
librealsense2-dkms
librealsense2-gl
librealsense2-gl-dbg
librealsense2-gl-dev
librealsense2-net
librealsense2-net-dbg
librealsense2-net-dev
librealsense2-udev-rules
librealsense2-utils
librscalibrationtool
```

There was a warning about `apt-key` being deprecated when adding the repository, but I could install the librealsense2-... packages so the repo itself seems to be added ok.

MartyG-RealSense, 2023-04-20:
Hi @jomiham  I will refer your question to my Intel RealSense colleagues.  In the meantime, calibration can alternatively be performed using the RealSense Viewer tool by going to the **More** option near the top of the Viewer's options side-panel and selecting **On-Chip Calibration** from its drop-down menu.

Intel's guide for On-Chip and other forms of calibration under the More menu is at the link below.

https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras

jomiham, 2023-04-20:
Thanks @MartyG-RealSense,

We currently do maintenance calibration using the `Intel.Realsense.DynamicCalibrator` in combination with the image from here:
https://dev.intelrealsense.com/docs/dynamic-calibration-print-target

How would the self-calibration process using the Viewer tool compare? Is it essentially just two interfaces to the same calibration or would one be better than the other?

Can you please keep me updated on the response regarding the missing package as we don't want to change calibration procedures unnecessarily.

MartyG-RealSense, 2023-04-20:
Self calibration and Dynamic Calibration are two different approaches to calibration that each have their own advantages and disadvantages.

Self-calibration is built into the camera firmware driver and so does not require the installation of a separate software package like the Dynamic Calibration tool does.  

Self-calibration only calibrates depth, whilst Dynamic Calibration calibrates both depth and RGB.

Self-calibration provides a Health Check score so that you can see at a glance how healthy or unhealthy the current calibration is.

The self-calibration toolset can calibrate depth quality (On-Chip), depth measurement accuracy (Tare) and Focal Length.

Yes, I will keep you updated about the response.

MartyG-RealSense, 2023-04-21:
I have received information from my RealSense colleagues.  The latest version of the Dynamic Calibration tool (2.13.1.0) has been published for Windows but the Linux version is still being published and will hopefully be done next week.  Apologies for confusion caused by the tool's download page information.

If your calibration need is urgent then we can share the Debian package so that you can install it manually.  The package size is about 120 MB.

jomiham, 2023-04-21:
Yes, if you can share the .deb package, that would be helpful. 
Thank you for the update!

MartyG-RealSense, 2023-04-21:
You are very welcome.  I will coordinate with my colleagues to share the package.

MartyG-RealSense, 2023-04-22:
You can download the Debian package from the link below by left-clicking on the download button on the top corner of the download page.

https://drive.google.com/file/d/1YQiXe0WXp45VW3Ln2dQpVYv8uyKWJXB4/view?usp=sharing

[image]

**To install:**
sudo dpkg -i librscalibrationtool_2.13.1.0_amd64.deb

**To list where the files installed:**
sudo dpkg -L librscalibrationtool

MartyG-RealSense, 2023-04-28:
I have been informed that the 22.04 version of the 2.13.1.0 calibration tool for Linux has now been published and can be installed with the commands below.

```
sudo apt update
sudo apt install librscalibrationtool
```

MartyG-RealSense, 2023-05-03:
Hi @jomiham  Do you require further assistance with this case, please?  Thanks!

jomiham, 2023-05-03:
Thanks @MartyG-RealSense,

I can see `librscalibrationtool` package now so you can close this issue.

closed 2023-05-03 by jomiham

MartyG-RealSense, 2025-03-12:
Hi @ruffsl  It looks as though you are searching for packages on the librealsense repository that contain the number '2', such as **librealsense2-dkms**. The 'librscalibrationtool' package - which is retrieved with `sudo apt-get install librscalibrationtool` - does not contain '2' in its name, which would likely exclude it from the search results.

Is the package findable if you change your filter command to the one below?

`curl https://librealsense.intel.com/Debian/apt-repo/dists/noble/main/binary-amd64/Packages 2>&1 | grep Package | awk '{print $calibration}' | sort -u`

fschiro, 2025-09-20:
I managed to spin it up with docker if anyone has this problem in the future:

```bash
xhost +local: && docker run -it --rm --privileged \
  -e DISPLAY=${DISPLAY} \
  -v /tmp/.X11-unix:/tmp/.X11-unix \
  -v /dev:/dev \
  --device /dev/bus/usb:/dev/bus/usb \
  --network host \
  ubuntu:22.04 bash -c "
    apt update && 
    apt install -y curl gnupg freeglut3-dev libgl1-mesa-glx libusb-1.0-0-dev && 
    curl -sSf https://librealsense.intel.com/Debian/librealsense.pgp | gpg --dearmor | tee /etc/apt/keyrings/librealsense.pgp > /dev/null && 
    echo 'deb [signed-by=/etc/apt/keyrings/librealsense.pgp] https://librealsense.intel.com/Debian/apt-repo jammy main' | tee /etc/apt/sources.list.d/librealsense.list && 
    apt update && 
    apt install -y librscalibrationtool && 
    Intel.Realsense.DynamicCalibrator
  "; xhost -local:
```

MartyG-RealSense, 2025-09-21:
Thanks so much @fschiro for sharing your approach!