realsenseai/librealsense #11715: GLFW Driver Error

morimio, 2023-04-20:
| | |
|---|---|
|**librealsense**|2.50.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
GLFW Driver Error: WGL: The driver does not appear to support OpenGL
Could not initialize offscreen context!

```

MartyG-RealSense, 2023-04-20:
Hi @morimio  This error can be resolved by installing or updating graphics drivers on the computer.

morimio, 2023-04-20:
Thank you for your reply.
However, same error comes everytime after installing.
Also, my computer is unable to update graphics drivers becase of my
computer's driver.
Could I have another way to resolve?

MartyG-RealSense, 2023-04-20:
Could you use the instructions at https://github.com/IntelRealSense/librealsense/issues/8024#issuecomment-747630656 to run an example called **rs-capture.exe** and let me know whether it works, please?

rs-capture requires OpenGL graphics like the RealSense Viewer does, so if rs-capture does not work then this would confirm that updating the computer's graphics drivers would be the only way to resolve this issue.

MartyG-RealSense, 2023-04-26:
Hi @morimio  Do you require further assistance with this case, please?  Thanks!

morimio, 2023-04-29:
Thank you very much. It helps a lot.

MartyG-RealSense, 2023-04-29:
You are very welcome.  Is your error now solved, please?  Thanks!

MartyG-RealSense, 2023-05-06:
Case closed due to no further comments received.

closed 2023-05-06 by MartyG-RealSense

yibochen38, 2024-03-04:
I solved this question  by updating the graphic driver.