realsenseai/librealsense #11717: Is the IMU on d455 realsense module not calibrated?

kumar-akshay324, 2023-04-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 5.14 | 
| Operating System & Version |   Linux (Ubuntu 20)| 
| Kernel Version (Linux Only)    |  5.10.104-tegra                                         | 
| Platform                                 | NVIDIA Jetson Xavier NX |
| SDK Version                            | 2.53.1| 
| Language                            |  C++                          | 

### Issue Description

Can you tell me the if IMU on the d455 module is calibrated or not? And if not, what are the steps to calibrate it in the field?

MartyG-RealSense, 2023-04-20:
Hi @kumar-akshay324  Yes, the IMU of the D455 camera is calibrated.  

A calibrated IMU has an option in the RealSense Viewer tool called **Enable Motion Correction** accessible and enabled by default in the 'Motion Module > Controls' category of the Viewer's options side-panel.  When enabled, this option performs correction on the IMU's raw data to make it more accurate by 'fixing' it.

[image]

If you wish to perform a re-calibration of the IMU then this can be done with a Python tool.

https://github.com/IntelRealSense/librealsense/tree/master/tools/rs-imu-calibration

A PDF user guide for the tool can be found at the link below.

https://dev.intelrealsense.com/docs/imu-calibration-tool-for-intel-realsense-depth-camera

kumar-akshay324, 2023-04-20:
To confirm, I am talking about the d455 module itself and not the camera package. We don't buy the fabricated but only the module. 

This is what we buy from you  - https://store.intelrealsense.com/buy-intel-realsense-depth-module-d450-and-d4-board-bundle-d.html

MartyG-RealSense, 2023-04-20:
The IMU is on the Vision Processor D4 V3 board and not the D450 Depth Module board.

https://www.framos.com/en/products/vision-processor-d4-v3-imu-vers-for-module-d450-25203

My understanding is that the product is uncalibrated and that calibration should be performed by the user once the D4 board and depth module have been joined together with an interposer.

closed 2023-05-02 by MartyG-RealSense