realsenseai/librealsense #11718: not able to install pyrealsense2 on jetson agx orin

sidh1603, 2023-04-20:
HI, i am using ubuntu 20.04 and i tried to install pyrealsense2 on jetson agx orin ,but i am getting error

ERROR: Could not find a version that satisfies the requirement pyrealsense2 (from versions: none)
ERROR: No matching distribution found for pyrealsense2

MartyG-RealSense, 2023-04-20:
Hi @sidh1603  This error occurs if performing the installation instruction **pip install pyrealsense2** on a computer with an Arm architecture, such as Nvidia Jetson.  The pyrealsense2 wrapper can only be installed with pip install on computers with x86 / x64 architecture, such as desktop and laptop PCs.  On Arm devices the wrapper must be built from source code.

Instructions for building librealsense and the wrapper together from source code on Jetson boards can be found at https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049

sidh1603, 2023-04-24:
hi,i have followed all the steps , but still no module pyrealsense2 , i am using python 3.8 , do i need to upgrade python??  or python3.8 does not support pyrealsense2??Sent from my iPhone

Instructions for building librealsense and the wrapper together from source code on Jetson boards can be found at #6964 (comment)

MartyG-RealSense, 2023-04-30:
Hi @sidh1603  Do you require further assistance with this case, please?  Thanks!

ronbitonn, 2023-05-02:
Hey @sidh1603  I have the same issue.
If you solve the problem please let me know what are the steps to the solution.

MartyG-RealSense, 2023-05-12:
Hi @sidh1603  Do you require further assistance with this case, please?  Thanks!

@ronbitonn  Were you able to solve your own Orin problem, please?  If you still require assistance then please create a new support ticket by visiting the link below to go to the front page of this support forum and clicking the green **New Issue** button.  Thanks!

https://github.com/IntelRealSense/librealsense/issues

ronbitonn, 2023-05-12:
@MartyG-RealSense 
Hello, thanks for the guidelines at [#6964](https://github.com/IntelRealSense/librealsense/issues/6964).
For now, it's working great.
Have a nice weekend.

MartyG-RealSense, 2023-05-12:
You are very welcome, @ronbitonn - I'm pleased to hear that you were successful.  :)

closed 2023-05-25 by MartyG-RealSense