realsenseai/librealsense #11720: intel realsense viewer wrong

rr922018, 2023-04-21:
| | |
|---|---|
|**librealsense**|2.53.1 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
GLFW Driver Error: WGL: The driver does not appear to support OpenGL
Could not initialize offscreen context!

```

MartyG-RealSense, 2023-04-21:
Hi @rr922018  This message indicates that the Windows computer's graphics drivers may be missing or outdated.  It is therefore recommended that the computer's graphics drivers be updated in order to resolve this error.

rr922018, 2023-04-21:
Thank you very much. I reinstalled the driver and now the problem is solved.

MartyG-RealSense, 2023-04-21:
You are very welcome.  It's great to hear that a driver reinstall resolved your problem.  Thanks for the update!

closed 2023-04-26 by MartyG-RealSense