realsenseai/librealsense #11722: Polling leads to too many file handles open

brunovollmer, 2023-04-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D430 & D430i | 
| Firmware Version                   | 05.14.00.00 | 
| Operating System & Version |  Linux, Ubuntu 22.04 | 
| Kernel Version (Linux Only)    |  5.10.66                                   | 
| Platform                                 | Khadas Edge 2  |
| SDK Version                            |  2.50.0       & 2.53.0                 | 
| Language                            |  Python                        | 
| Segment			|  Robotics                    | 

### Issue Description

We are working with a multi-camera setup that consists of two D430 and one D430i. We are using python for our image acquisition pipeline and have switched to `poll_for_frames` instead of `wait_for_frames` to support the different FPS of the IMU and the depth + greyscale pipeline.

The problem we are encountering right now is that after running the acquisition pipeline for a while the application crashes with either of the two errors.

```
File "<string>", line 2, in __getitem__
File "/usr/local/lib/python3.8/multiprocessing/managers.py", line 835, in _callmethod
File "/usr/local/lib/python3.8/multiprocessing/connection.py", line 251, in recv
File "/usr/local/lib/python3.8/multiprocessing/managers.py", line 959, in RebuildProxy
File "/usr/local/lib/python3.8/multiprocessing/managers.py", line 809, in __init__
File "/usr/local/lib/python3.8/multiprocessing/managers.py", line 863, in _incref
File "/usr/local/lib/python3.8/multiprocessing/connection.py", line 502, in Client
File "/usr/local/lib/python3.8/multiprocessing/connection.py", line 628, in SocketClient
File "/usr/local/lib/python3.8/socket.py", line 231, in __init__
OSError: [Errno 24] Too many open files
```

```
python: ../../libusb/io.c:2146: handle_events: Assertion `ctx->pollfds_cnt >= internal_nfds' failed.`
```

We have reduced it to a simple recording script that shows the same problems. After some investigation, it seems that the script for whatever reason opens more and more file handles without closing the previous ones. 

[image]

This leads to a crash at one point as Linux limits the amount of allowed open file handles. 

We are now wondering if you know what could cause such a problem and if there is a way to release those file handles correctly. We have tested the script with 2.50 and 2.53. Additionally, we have compiled librealsense with RSUSB, ss we are using a not supported kernel.

MartyG-RealSense, 2023-04-22:
Hi @brunovollmer  poll_for_frames() is the appropriate method to use in a multiple camera application instead of wait_for_frames().

The official pyrealsense2 documentation states about poll_for_frames: "The application can **maintain the frames handles to defer processing**.  However, if the application maintains too long history, the device may lack memory resources to produce new frames, and the following calls to this method shall return no new frames, until resources become available".

https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.pipeline.html

brunovollmer, 2023-04-23:
I've seen that. I've checked and I haven't been able to figure out what could cause the frames to be maintained at all. What I'm doing right now is to process the frame right after before polling again. 

Additionally, but that is more a general concern, the Python docs you are referencing are from version 2.33. Would it be possible to update that from the intel side?

MartyG-RealSense, 2023-04-23:
The main official documentation site that is regularly updated is the **Intel RealSense Cross Platform API** one.  It has similar descriptions to the pyrealsense2 documentation.  Its poll_for_frames entry is here:

https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1pipeline.html#a9069a979fd9e1e28881b945a6aec1b79

Because pyrealsense2 is a wrapper of the C++ library, every C++ instruction should automatically have a pyrealsense2 equivalent.

closed 2023-05-06 by MartyG-RealSense