realsenseai/librealsense #11723: The difference of param about LeftRigth and RGB in D405

hfdlwt, 2023-04-23:
Hi！ I would like to update the calibration param to RealsenseD405 ， But I dont know how to deal with the RGB param with Intel.Realsense.CustomRW tool ？ Should I only need to update the parameters of the left and right cameras?  What is the difference between param about LeftRigth and RGB in D405？
[image]

And why cant subscribe the infra2 camera_info from rostopic and  it is same with the infra1
[image]

[image]

MartyG-RealSense, 2023-04-23:
Hi @hfdlwt  A case from earlier this week at https://github.com/IntelRealSense/librealsense/issues/11693 where another RealSense user was attempting to configure D405 RGB calibration values with CustomRW may be a useful reference.

If you are not using the RGB stream of D405 - which is provided by the depth sensor as D405 does not have a separate RGB sensor - then you should not need to be concerned about RGB calibration values.  If an RGB sensor is not detected then the calibration tool just calibrates depth.

********

Infra1 and Infra2 are disabled in the RealSense ROS wrapper by default and can be published by setting them to True in the launch file or by adding the commands below onto the end of the roslaunch instruction:

**enable_infra1:=true enable_infra2:=true**

The Infra2 topic will only be published if the camera is on a USB 3 connection.  On a USB 2 connection only Infra1 is available.

**********

Page 12 of the PDF user guide document for the Dynamic Calibration tool provides descriptions of the extrinsic parameters.

https://www.intel.com/content/www/us/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html

[image]

hfdlwt, 2023-04-23:
@MartyG-RealSense Thanks for your quickly replay!But I use the USB 3.2 connection there is still no info about infra2.
[image]

MartyG-RealSense, 2023-04-23:
The infrared streams in the Viewer are only selectable when the Stereo Module is disabled (a red icon beside it).  The Infrared and Infrared 2 boxes should be enabled by left-clicking on the black box beside them to turn the box blue (On).

[image]

hfdlwt, 2023-04-23:
@MartyG-RealSense RealSense OK,thanks！another question ， how can i get the infra2/right camera param and to calibrate the infra2/right camera?

MartyG-RealSense, 2023-04-23:
Infra2 should be published in the ROS wrapper if the camera is detected as being on a USB 3 connection and **enable_infra2** is true.

If you have enabled Infra2 in the ROS wrapper but the Infra2 topic is not published then you can check in the first half of the launch log for a line called **Device USB type**.  If it has 3.2 beside it then the camera is detected as being on a USB 3 connection.  If it has 2.1 beside it then the camera is being mis-detected by the ROS wrapper as being on a USB 2 connection, preventing Infra2 from being published.

MartyG-RealSense, 2023-05-04:
Hi @hfdlwt  Do you require further assistance with this case, please?  Thanks!

My Intel RealSense colleagues confirmed to me that RGB on the D405 camera model is calibrated automatically when depth is calibrated because the RGB uses the same left IR sensor that depth originates from.

closed 2023-05-10 by MartyG-RealSense