realsenseai/librealsense #11724: intel realsense viewer cannot work

VirtuousContract, 2023-04-23:
| | |
|---|---|
|**librealsense**|2.53.1 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Unknown error (-529697949)!
```

MartyG-RealSense, 2023-04-23:
Hi @VirtuousContract  Does the error appear when the RealSense Viewer first opens, or does it appear when an option is selected?

For example, this error sometimes occurs when trying to access the RGB camera from the **Add Source** option, like in https://github.com/IntelRealSense/librealsense/issues/11338

VirtuousContract, 2023-04-23:
Thank you for your reply. This issue occurs every time the viewer is opened. I suspect it may be due to the data cable. I plan to repurchase the data cable to verify it.

MartyG-RealSense, 2023-04-23:
Thanks very much for the update.  If you are using your own choice of USB cable, problems may occur if (a) the cable length is longer than 2 meters and the cable is a low-quality one or one that is designed for charging instead of data transfer; (b) if the cable has internal damage, perhaps from being bent a lot; or (c) if the cable is a USB C to C instead of a Type-C (USB A to C).

closed 2023-05-08 by MartyG-RealSense