realsenseai/librealsense #11725: How can I ignore camera point?

qq996235826, 2023-04-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | (05.14.00.00) | 
| Operating System & Version |   Win 11| 
| Platform                                 | PC |
| SDK Version                            |  2.53.1                         | 
| Language                            |  python                         | 
| Segment			|  Robot                        | 

### Issue Description
The point cloud I got seems to carry the point of the camera's own position, but I don't want this point. How can I get the point without the camera's position？
I use point clouds for plane fitting, but sometimes camera points affect my fitting results, which makes my results unreliable, so I want to discard camera points.
[image]

[image]

this is my code
```
            self.pc.map_to(color_frame)
            points = self.pc.calculate(temp_filtered)
            # save ply point cloud
            points.export_to_ply(config.final_path+'point_cloud.ply', color_frame)
            vtx = np.asanyarray(points.get_vertices())
            colorful = np.asanyarray(color_frame.get_data())
            colorful = colorful.reshape(-1, 3)
            point_cloud = []
            index = 0
            for p in vtx:
                point_cloud.append(
                    [p[0], p[1], p[2], colorful[index][0], colorful[index][1], colorful[index][2]])
                index += 1
            points.export_to_ply(config.BASE_CLOUD_PATH, color_frame)
            np.save(config.NPY_PATH, point_cloud)
```

MartyG-RealSense, 2023-04-23:
Hi @qq996235826  Using the RealSense SDK instruction **rs2_transform_point_to_point** to transform points from one coordinate space may be what you are seeking.  More information about this can be found at https://github.com/IntelRealSense/librealsense/issues/5583

qq996235826, 2023-04-26:
> Hi @qq996235826 Using the RealSense SDK instruction **rs2_transform_point_to_point** to transform points from one coordinate space to another may be what you are seeking. More information about this can be found at #5583

I had trouble using rs2_transform_point_to_point，But I used open3d's remove_statistical_outlier method to remove the point

closed 2023-04-26 by qq996235826

MartyG-RealSense, 2023-04-26:
I'm pleased to hear that you achieved a solution.  Thanks very much for the update!