realsenseai/librealsense #11727: Message was lost

shivangi-futuristicbots, 2023-04-24:
HI @MartyG-RealSense, 
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.14.0 | 
| Operating System & Version |   Linux  | 
| Platform                                 | Jetson 5 xavior  |
| SDK Version                            |  2.0                  | 
| Language                            |  python                        | 

### Issue Description
I use the [#10416](https://github.com/IntelRealSense/librealsense/issues/10416#issuecomment-1427004251) issue to install the SDK of the realsense camera D435i, but i am able to run the `realsense-viewer` command but the application is not started.
I connect my PC in the same network as jetson to get the ros2 topic list which i am able to fetch the when i echo the topic it givinng the 
[image]
i try export ROS_DOMAIN_ID=80 on both device but not worked.
Could you help me to resolve this?

i read the [Building the source](https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras)
which give the additional command to run the **-DBUILD_NETWORK_DEVICE=ON** & **-DFORCE_RSUSB_BACKEND=ON**  to build the package so, i replace the cmake command: 

from` cmake ../ -DFORCE_RSUSB_BACKEND=false -DBUILD_PYTHON_BINDINGS=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true` 
to `cmake ../ -DBUILD_NETWORK_DEVICE=ON-DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true`
run the make command inside the build folder

MartyG-RealSense, 2023-04-24:
Hi @shivangi-futuristicbots  Is the realsense-viewer tool able to open but does not show the camera, or does the realsense-viewer window not appear at all?

shivangi-futuristicbots, 2023-04-24:
i am not able to see the viewer window  at all

MartyG-RealSense, 2023-04-24:
The build flag **-DBUILD_NETWORK_DEVICE=ON** builds support in the SDK for accessing another computer over ethernet by logging in to the second computer (the one that the camera is attached to) using an **ssh** connection and connecting to the **rs-server** networking tool.  This networking system cannot be used for ROS networking by default, though it is possible to modify it to use ROS networking protocols instead of the RTSP protocol that the rs-server system uses.

In an rs-server system, realsense-viewer should be accessed from the 'host' computer (the one that the camera is not attached to).  My understanding is that the SDK should also be built with the **-DBUILD_NETWORK_DEVICE=ON** flag on the camera-less host computer too

If you would like to access the camera on another computer using ROS, Intel have a guide at the link below for doing so.

https://github.com/IntelRealSense/realsense-ros/wiki/showcase-of-using-3-cameras-in-2-machines

shivangi-futuristicbots, 2023-04-26:
Hi @MartyG-RealSense, i tried to resolve it but i am not able to get **/camera/depth/image_rect_raw** and  **/camera/color/image_raw** topic information from Jetson 5 xavior NX to my PC, could you suggest me more better solutions.

MartyG-RealSense, 2023-04-26:
The guide below for communicating with ROS over networking between a Jetson and a PC may provide some ideas.

https://risc.readthedocs.io/2-ros-network-wifi-ethernet.html

closed 2023-05-08 by MartyG-RealSense