realsenseai/librealsense #11729: realsense L515 haneye

zhanglepy, 2023-04-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

zhanglepy, 2023-04-25:
How to perform hand eye calibration on the realsense L515 camera, with the eyes in hand？

MartyG-RealSense, 2023-04-25:
Hi @zhanglepy  There is information about hand-eye calibration with L515 at https://github.com/IntelRealSense/librealsense/issues/10580#issuecomment-1156190607 that may be helpful to you.

zhanglepy, 2023-04-26:
Hi @MartyG-RealSense Thank you for your reply. I have one problem. Which depth camera is suitable for obtaining RGB-D information of point clouds at close range (with a distance of less than 0.3m and an accuracy of about 1mm)? For example, which is the most suitable way for structured light, lidar and binocular depth camera? If using a realsense depth camera, which one is suitable, D435i, D455, or L515?

MartyG-RealSense, 2023-04-26:
I have replied to the above question at https://github.com/IntelRealSense/librealsense/issues/11732

MartyG-RealSense, 2026-10-23:
This open GitHub issue relating to the RealSense L515 camera is being closed because the L515 model is End of Life (EOL) and no longer supported.  We welcome the creation of new issues relating to supported cameras.

closed 2026-10-23 by MartyG-RealSense