realsenseai/librealsense #11730: Dynamic Calibration Tool using network device

RobaczeQ, 2023-04-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D410 | 
| Firmware Version                   | 2.13.1.0 | 
| Operating System & Version |  Win 11  | 
| Kernel Version (Linux Only)    |  -                                    | 
| Platform                                 | PC  |
| SDK Version                            |  v2.53.1                         | 
| Language                            |  C/opencv/Matlab                          | 
| Segment			|  Robot/VR/AR/                       | 

### Issue Description
I have Intel® RealSense™ Depth Module D410 embedded into Kinova gen3 robotic arm and was trying to get Dynamic Calibration Tool to work but there is "No device detected" and there is no add button for network device/source. 
In RealSense Viewer there is add source "Add network device" when it is compiled with CMake flag -DBUILD_NETWORK_DEVICE=ON. But I don't see such feature/flag in Calibration Tool. 
Is there any workaround when I can't connect camera thru USB?

MartyG-RealSense, 2023-04-25:
Hi @RobaczeQ  Products with integrated RealSense cameras that are not USB connected typically have interfaces provided by the product's manufacturer instead of using Intel's RealSense SDK.  

With products that have an integrated RealSense camera but the overall product is not manufactured by Intel, the manufacturer is responsible for technical support of that product. 

Page 117 of the Kinova gen3 user manual describes how its Run mode can be set to **Calibration** using the Kinova API or web app.

https://www.kinovarobotics.com/uploads/User-Guide-Gen3-R07.pdf

[image]

RobaczeQ, 2023-04-25:
> ________
> # enum CalibrationItem
> 
> ## Overview / Purpose
> 
> Enumeration CalibrationItem
> 
> |Enumerator|Value|Description|
> |----------|-----|-----------|
> |UNSPECIFIED\_CALIBRATION\_ITEM|0|Unspecified calibration item|
> |COGGING|1|Cogging calibration|
> |MAGNETIC|2|Magnetic sensors calibration|
> |MOTOR|3|Motor calibration|
> |POSITION\_RANGE|4|Position range calibration|
> 
> _____
There is no calibration for camera. So, there is no easy way for add network device if I can access it through rtsp?

MartyG-RealSense, 2023-04-25:
The calibration tools do not integrate with the ethernet networking system that is built with -DBUILD_NETWORK_DEVICE=ON. 
 The networking system is also being removed in the next version of the RealSense SDK after the current one (2.53.1).  This is because it was an experimental proof of concept rather than an official, validated Intel RealSense feature.

The user guide for the Dynamic Calibration tool states in section **4.3  Connect camera to host system** that whilst the camera that is to be calibrated will usually be connected to the host computer via a USB cable, "on systems with camera integrated it's already connected, so no extra step is required".

https://www.intel.com/content/www/us/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html 

This implies that the Dynamic Calibration tool should be able to detect an integrated camera if the tool is installed onto a computer without having to install ethernet networking.  The tool supports Windows and Ubuntu.

Alternatively, if the $1500 USD **OEM Calibration Target** product at the link below is purchased then it comes with a special OEM version of the Dynamic Calibration tool that can connect remotely to a camera via an Ubuntu server.  I could not guarantee that it would be able to connect to the integrated camera on the robot arm however.

https://store.intelrealsense.com/buy-intel-realsense-d400-cameras-calibration-target.html

closed 2023-04-25 by RobaczeQ