realsenseai/librealsense #11732: Which depth camera is suitable for obtaining RGB D information of point clouds at close range (with a distance of less than 0.3m and an accuracy of about 1mm)? For example, which is the most appropriate way to use structured light, lidar and binocular depth camera? If using a realsense depth camera, which one is suitable, D543i, D455, or L515?

zhanglepy, 2023-04-26:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

zhanglepy, 2023-04-26:
Which depth camera is suitable for obtaining RGB D information of point clouds at close range (with a distance of less than 0.3m and an accuracy of about 1mm)? For example, which is the most suitable way to use structured light, lidar and binocular depth camera? If using a realsense depth camera, which one is suitable, D543i, D455, or L515?

MartyG-RealSense, 2023-04-26:
Hi @zhanglepy  I can only advise about RealSense camera products.  In the RealSense range, the model that will likely best fit your needs is the D405.  It is a short-range camera (ideal depth range 7 cm to 50 cm / 0.5 m) that provides high quality, high accuracy images at close range.

https://store.intelrealsense.com/buy-intel-realsense-depth-camera-d405.html

zhanglepy, 2023-04-27:
Thank you for your reply!

MartyG-RealSense, 2023-05-02:
Hi @zhanglepy  Do you require further assistance with this case, please?  Thanks!

zhanglepy, 2023-05-02:
Based on your recommendation, I have purchased a D405 camera. After a simple test, I found that the distance and position accuracy fluctuate greatly. Can you recommend some cases of developing a D405 camera to me? Mainly used to obtain high-precision point clouds and 3D reconstruction of point clouds. Thank you!

MartyG-RealSense, 2023-05-02:
The D405 camera is not equipped with the infrared dot pattern projector that other RealSense 400 Series camera models have, so it can have difficulty in analyzing smooth, low-detail surfaces for depth information.  Increasing the light illumination in the location that the camera is being used in may help if it is possible to do so, as 400 Series cameras can alternatively use the ambient light in a location for depth analysis instead of the dot pattern projection.

Projecting a visible pattern onto surfaces in the scene to add texture to smooth surfaces should also help.  Visible light projection is discussed in the section linked to below of Intel's guide to use of projectors with 400 Series cameras.

https://dev.intelrealsense.com/docs/projectors#6-the-consumer-front-projector

 If you are using the RealSense Viewer tool then depth fluctuations can be reduced by expanding open the settings of the post-processing **Temporal Filter** and changing its value from the default of '0.4' to the lower value of '0.1'.

[image]

Going to **Advanced Controls > Depth Control** and changing the 'DS Second Peak Threshold' option from '325' to '0' may also help stabilization.

[image]

zhanglepy, 2023-05-05:
Hi，@MartyG-RealSense  Through your suggestion, the problem has indeed improved, but I do not want to adjust parameters through the RealSense Viewer tool. I want to adjust parameters through the Python librealsense library to obtain high-precision, clear, and stable deep point cloud information. Thanks!

zhanglepy, 2023-05-05:
Hi，@MartyG-RealSense The Python program I found on the official website doesn't seem to be suitable for D405. Can you help me recommend some Python programs developed for D405?

MartyG-RealSense, 2023-05-05:
Hi @zhanglepy  If you are referring to the depth-color alignment example **align_depth2color.py**, it fails on D405 because it checks for the presence of an RGB sensor, which the D405 camera model does not have as RGB is provided from its depth sensor.  The script can be edited to remove the check for an RGB sensor so that the script will work on D405.  A RealSense user shares their edit for doing so at https://github.com/IntelRealSense/librealsense/issues/11329