realsenseai/librealsense #11735: librealsense2-utils packets not found

MateuszSoczalski, 2023-04-26:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |     Linux (Ubuntu 22)                              | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Raspberry Pi                                          |
| SDK Version                            |  2                                                          | 
| Language                            |  {C/C#python                                           | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Hello, I've been trying to install the SDK but i keep getting stuck on these problems and cant find a way around it. It is the same with dkms dev and dbg packets.
Also there is a problem with this command:
"sudo add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo $(lsb_release -cs) main" -u"
It cant find the repository.>

MartyG-RealSense, 2023-04-27:
Hi @MateuszSoczalski  DKMS package installations are for x86 / x64 architecture computers such as desktop and laptop PCs.  For the Raspberry Pi, which has an Arm architecture, it is recommended that the SDK is built from source code with the **RSUSB backend** installation method described at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692

MateuszSoczalski, 2023-04-27:
Thank you for replying, i will try this as soon as i can.

MateuszSoczalski, 2023-04-28:
I tried to use the method but after running step 5 the Raspberry freezes at Performing tests, exactly at "Performing Test HAVE_STRUCT_TIMEVAL"

MartyG-RealSense, 2023-04-28:
On a low-end computing device, using a '-j' number in the make command such as **make -j8** can cause a random crash during build.  Please try `sudo make uninstall && make clean && make && sudo make install` without the -j8 parameter (which will have the side-effect of a longer build time) or a lower j number such as -j4.

MartyG-RealSense, 2023-05-10:
Case closed due to no further comments received.

closed 2023-05-10 by MartyG-RealSense

MateuszSoczalski, 2023-05-12:
Hi, sorry for lack of updates Installation was successful, but when i try to run a code that retrieves distance from the d435 camera i get the following error: pipeline.start(config)
RuntimeError: No device connected

MartyG-RealSense, 2023-05-12:
I'm pleased to hear that your installation was successful.

Is the camera detected if you try pre-made RealSense SDK example programs and tools like the ones located in Ubuntu's **usr/local/bin** folder (such as rs-capture and realsense-viewer).

MateuszSoczalski, 2023-05-15:
Yes, realsense viewer is working fine.

MartyG-RealSense, 2023-05-15:
Can you link me to the distance code that you tried or paste the code in a comment below if it is a simple script, please?

MateuszSoczalski, 2023-05-15:
Here it is:
import pyrealsense2 as rs

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 0, 640, 480, rs.format.z16, 30)

pipeline.start(config)

try:
    while True:
        frames = pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        if not depth_frame:
            continue

        width = depth_frame.get_width()
        height = depth_frame.get_height()
        x = width // 2
        y = height // 2
        dist = depth_frame.get_distance(x, y)

        print("Distance at center pixel: {} meters".format(dist))

finally:
    pipeline.stop()

Also i learned recently that the code doesnt work while realsense viewer is running.

MartyG-RealSense, 2023-05-16:
If a stream is enabled in one program then a 'claim' is placed on it that prevents it from being used by another program until the first program stops streaming.  So if your script is run first then the RealSense Viewer cannot use the stream, and if your script is run first then the Viewer cannot use the stream.

The Viewer does not need to be running in order for a script to work.