realsenseai/librealsense #11736: Accuracy of hand eye calibration (eye in hand) for D400 series cameras

zhanglepy, 2023-04-27:
Accuracy of hand eye calibration (eye in hand) for D400 series cameras

Recently, I was using Halcon to perform hand eye calibration on the D435i camera (with eyes in hand), and the results showed an average distance error of 1.135mm and an average angle error of 0.659 degrees. Is this error significant? May I ask how much calibration error you have, what other methods or software are used, and what methods can be used to reduce calibration error? Could you please introduce the hand eye calibration accuracy of D405 and D435i (with eyes in hand)? I would like to compare them. Thank you!

MartyG-RealSense, 2023-04-27:
Hi @zhanglepy  The D435i has less than 2% error at a distance of 2 meters according to the official camera model comparison matrix at the link below.

https://www.intelrealsense.com/compare-depth-cameras/

There are not accuracy values available specifically for hand-eye calibration.

Depth error starts at zero at the camera lenses and increases linearly over distance.  This is called **RMS Error**.  The orange line on the chart below predicts how error (mm) will increase over distance (m) on a D435 / D435i camera model.

[image]

For example, on that orange scale the depth error in mm at 1 m distance from the camera is predicted to be between 2.5 and 5 mm.

*******

There are hand-eye calibration tools for robot arms that work with RealSense 400 Series cameras.  Here are a couple of suggestions.

**autoCalibration**
https://github.com/IntelRealSense/librealsense/issues/3569#issuecomment-475621533 

**'Robotic Hand-eye Calibration Workspace' eye-in-hand for ROS**
https://github.com/lixiny/Handeye-Calibration-ROS 

The above system makes use of **visp**, which has been separately recommended by some for hand-eye calibration with RealSense.  http://wiki.ros.org/visp_hand2eye_calibration 

Another example of a hand-eye calibration tool that can be used with RealSense is **MoveIt**.

https://github.com/ros-planning/moveit_calibration

*********

You can find references about reducing hand-eye calibration error by googling for the search term below, including the quotation marks to make the search results relevant.

**"hand-eye calibration" "reduce error accumulation"**

MartyG-RealSense, 2023-05-08:
Case closed due to no further comments received.

closed 2023-05-08 by MartyG-RealSense

Yiiipu, 2024-02-05:
I also have poor repeatability (>1mm). Have you solved your problem?