realsenseai/librealsense #11739: HDR issue on Android

electro-logic, 2023-04-27:
Hello,
I'm trying HDR on Android, but I get a flickering image by using the following logic

var sensor = device.QuerySensors()[0];
sensor.SetValue(Option.HdrEnabled, 1);
sensor.SetValue(Option.SequenceSize, 2);
sensor.SetValue(Option.SequenceId, 1);

var hdr = new HdrMerge();
var frames = hdr.Process(_pipeline.WaitForFrames())
_glSurfaceView1.Upload(frames);

I get only even frame numbers but they are flickering, looks like something is not working on the HdrMerge

MartyG-RealSense, 2023-04-27:
Hi @electro-logic  Is it the infrared that is flickering, please?

If it is then https://github.com/IntelRealSense/realsense-ros/issues/2667#issuecomment-1506867003 lists details of the settings that stop infrared flickering when using HDR.  Comparing your settings to the Viewer's, it looks as though you have not set **SequenceName** to '1'.  

The post-processing filters 'HDR Merge' and 'Filter By Sequence ID' should also be set up and applied, with Filter By Sequence ID set to a value of '1'.

Metadata support should also be enabled.  A camera without metadata support will have flickering infrared even if all the settings and filters are configured correctly.

electro-logic, 2023-04-27:
Hello,

Now I have:

config.EnableStream(StreamType.Color, -1, IMAGE_WIDTH, IMAGE_HEIGHT, StreamFormat.Rgb8, 6);
config.EnableStream(StreamType.Depth, -1, IMAGE_WIDTH, IMAGE_HEIGHT, StreamFormat.Z16, 6);
config.EnableStream(StreamType.Infrared, -1, IMAGE_WIDTH, IMAGE_HEIGHT, StreamFormat.Y8, 6);    // Infrared-1
_pipeline.Start(config);

var sensor = device.QuerySensors()[0];
sensor.SetValue(Option.HdrEnabled, 1);
sensor.SetValue(Option.SequenceSize, 2);
sensor.SetValue(Option.SequenceId, 1);
sensor.SetValue(Option.SequenceName, 1);

var seq = new SequenceIdFilter(1);
var hdr = new HdrMerge();
var frames = seq.Process(hdr.Process(_pipeline.WaitForFrames()))
_glSurfaceView1.Upload(frames);

but IR and Depth frames are still flickering. I haven't found a way to explicitly enable metadata on Android.

MartyG-RealSense, 2023-04-27:
On Windows, metadata can be enabled in the RealSense Viewer with a pop-up box that appears when the Viewer is launched.  

Alternatively it can be enabled by editing the registry of Windows, though this should not be attempted unless familiar with registry edting.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md#enabling-metadata-on-windows

On Linux, metadata support is enabled automatically if the SDK is built from Debian packages or built from source code with the **RSUSB Backend** installation method. 

When building the SDK from source code but not using the RSUSB installation method, it is enabled by applying a kernel patch.

electro-logic, 2023-04-27:
Hello,
I am using 1280x720@6fps.
I think Metadata are automatically enabled, because I can get the SequenceID and the Frame Number.
I'm back to RealSenseViewer to experiment a bit more with HDR. How can I specify exposure for the dark frame? By default is too low (ex. Exposure=1) and almost nothing is visible.

MartyG-RealSense, 2023-04-27:
When HDR's sequence options (Size, Name, Id) have a value set for them, auto-exposure is automatically disabled.  This is referenced in the Limitations section of the HDR white-paper guide.

https://dev.intelrealsense.com/docs/high-dynamic-range-with-stereoscopic-depth-cameras#43-limitations

electro-logic, 2023-04-27:
Hello Marty, thank you very much for the support.
I'm dropping the HDR feature in my app because there are too many limitations.
Would be useful to have something similar to modern camera, where HDR can be configured as +/- 2 stops, where an 0 stop is the auto exposure.

electro-logic, 2023-05-04:
Can be closed, thank you

closed 2023-05-04 by electro-logic