realsenseai/librealsense #11740: Android GLRsSurfaceView not showing a single Frame

electro-logic, 2023-04-27:
Hello,

This code is working

var frames = _pipeline.WaitForFrames().ReleaseWith(fr).As(Extension.Frameset) as FrameSet
_glSurfaceView1.Upload(frames);

But if I want to display only the Depth frame

var depthFrame = frames.First(StreamType.Depth).ReleaseWith(fr).As(Extension.DepthFrame) as DepthFrame
_glSurfaceView1.Upload(depthFrame);

this code is not showing anything. Is this a bug?

MartyG-RealSense, 2023-04-28:
Hi @electro-logic  Are you able to achieve the depth stream only if you define a custom **config** stream configuration like in https://github.com/IntelRealSense/librealsense/issues/11670 so that only the depth stream is enabled and the color stream is not?  If config instructions are not defined then the SDK will enable both the depth and color streams by default.

MartyG-RealSense, 2023-04-28:
I wonder whether you could configure two separate pipelines, each with their own config instructions, and put depth on 'pipeline1' and color + IR on 'pipeline2'.  https://github.com/IntelRealSense/librealsense/issues/1735#issuecomment-390840337 has an example of this method in Python whose structure should be convertable to Android instructions, since you know how to define a pipeline and a config instruction.

Having set up individual pipelines, you would then have two pipeline start lines for 'pipeline1' and 'pipeline2'.

electro-logic, 2023-04-28:
I need frames to be synchronized and, though may work, this looks like a workaround. Any chance to get this fixed?

MartyG-RealSense, 2023-04-29:
I am not aware of a past example of a RealSense Android case in which the visual rendering of one stream type has successfully been disabled whilst the other is enabled.  So I do not have al alternative suggestion, unfortunately.

electro-logic, 2023-04-29:
I can always process the depth data and create a bitmap without the GLRsSurfaceView, I will try this way.

electro-logic, 2023-04-29:
Hello, at the end I solved in this way (pseudocode)

```
byte[] depthData = new byte[IMAGE_WIDTH * IMAGE_HEIGHT * 2];
int[] depthDataARGB = new int[IMAGE_WIDTH * IMAGE_HEIGHT];
depthFrame.GetData(depthData);
Bitmap bmp = Bitmap.CreateBitmap(IMAGE_WIDTH, IMAGE_HEIGHT, Bitmap.Config.Argb8888, false);

// Convert Z16 to ARGB888
for (int indexRgb24 = 0; indexRgb24 < IMAGE_WIDTH * IMAGE_HEIGHT; indexRgb24++)
{
    var offsetDepth = indexRgb24 * 2;                                
    short depth = (short)(depthData[offsetDepth] | (depthData[offsetDepth + 1] << 8));
    byte intensity = (byte)(depth / (double)short.MaxValue * 255.0);
    depthDataARGB[indexRgb24] = intensity | (intensity << 8) | (intensity << 16) | (255 << 24);
}
bmp.SetPixels(depthDataARGB, 0, IMAGE_WIDTH, 0, 0, IMAGE_WIDTH, IMAGE_HEIGHT);
_imageView1.SetImageBitmap(bmp);
```

Thanks anyway for the support

MartyG-RealSense, 2023-04-29:
No problem at all, @electro-logic - I'm pleased to hear that you achieved a solution!

closed 2023-05-04 by electro-logic