realsenseai/librealsense #11742: Unable to read IMU - D435i, Jetson Nano, Docker

EnthusiasticRoboticist, 2023-04-28:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.14.00.00 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.9.299-tegra                                         | 
| Platform                                 | NVIDIA Jetson Nano  |
| SDK Version                            |  2.53.1                          | 
| Language                            |  C++                          | 
| Segment			|  Robot                           | 

### Issue Description

I cannot read the IMU data appropriately from D435i on Jetson Nano in docker.

When I run the test code below multiple times, different outputs are produced (in order of frequency):

1. Sometimes it only outputs gyro frames
2. it outputs only 3 gyro frames and then nothing
3. it outputs nothing at all
4. it only outputs acceleration frames
5. Very rarely it outputs both acceleration and gyro data. In this case, it seems the frequency of both data types are almost equal. I thought the accel data should have been a lot slower (63Hz accel vs 200Hz gyro).

I have also tested with the realsense viewer in the same docker image. I observed the same behavior with realsense viewer on Jetson Nano. Realsense viewer on a desktop Ubuntu machine shows both Gyro and Accel but with a lot of frame drops (USB 2.1). My test code works fine on an X86 Ubuntu machine (I guess). I am using the same dockerfile for X86 but with a different base image. Here are the base images:
* `nvcr.io/nvidia/l4t-base:r32.5.0` and for Jetson Nano
* `nvcr.io/nvidia/cudagl:11.4.2-devel-ubuntu18.04` for X86.

[Here is my dockerfile that compiles the realsesne SDK](https://github.com/EnthusiasticRoboticist/DonkeyJet/blob/main/realsense.Dockerfile).

Depth data alone is fine.

Thanks

Here is my test code
```.cpp
TEST(realsense_test, imu_test)
{
  std::map<int, int> counters;
  std::map<int, std::string> stream_names;
  std::mutex mutex;

  {
    rs2::context ctx;
    auto list = ctx.query_devices();
    rs2::device dev = list.front();
    std::cout << "Device name: " << dev.get_info(RS2_CAMERA_INFO_NAME) << std::endl;
  }

  // Define frame callback
  // The callback is executed on a sensor thread and can be called simultaneously from multiple sensors
  // Therefore any modification to common memory should be done under lock
  auto callback = [&](const rs2::frame &frame)
  {
    std::lock_guard<std::mutex> lock(mutex);
    if (rs2::frameset fs = frame.as<rs2::frameset>())
    {
      std::cout << "nonIMU data" << std::endl;
      // With callbacks, all synchronized stream will arrive in a single frameset
      for (const rs2::frame &f : fs)
        counters[f.get_profile().unique_id()]++;
    }
    else
    {
      // Stream that bypass synchronization (such as IMU) will produce single frames
      counters[frame.get_profile().unique_id()]++;
      auto motion = frame.as<rs2::motion_frame>();
      if (motion && motion.get_profile().stream_type() == RS2_STREAM_GYRO &&
          motion.get_profile().format() == RS2_FORMAT_MOTION_XYZ32F)
      {
        rs2_vector gyro_data = motion.get_motion_data();
        std::cout << "gyro frame: " << gyro_data.x << ", " << gyro_data.y << ", " << gyro_data.z << std::endl;
      }

      if (motion && motion.get_profile().stream_type() == RS2_STREAM_ACCEL &&
          motion.get_profile().format() == RS2_FORMAT_MOTION_XYZ32F)
      {
        rs2_vector accel_data = motion.get_motion_data();
        std::cout << "accel frame: " << accel_data.x << ", " << accel_data.y << ", " << accel_data.z << std::endl;
      }
    }
  };

  // Declare RealSense pipeline, encapsulating the actual device and sensors.
  rs2::pipeline pipe;

  rs2::config rs_cfg_;
  rs_cfg_.enable_stream(RS2_STREAM_ACCEL, RS2_FORMAT_MOTION_XYZ32F);
  rs_cfg_.enable_stream(RS2_STREAM_GYRO, RS2_FORMAT_MOTION_XYZ32F);

  rs2::pipeline_profile profiles = pipe.start(rs_cfg_, callback);

  // Collect the enabled streams names
  for (auto p : profiles.get_streams())
    stream_names[p.unique_id()] = p.stream_name();

  std::cout << "RealSense callback sample" << std::endl
            << std::endl;

  while (true)
  {
    std::this_thread::sleep_for(std::chrono::seconds(1));

    std::lock_guard<std::mutex> lock(mutex);

    for (auto p : counters)
    {
      std::cout << std::endl;
      std::cout << stream_names[p.first] << "[" << p.first << "]: " << p.second << " [frames] || ";
    }
  }
}
```

MartyG-RealSense, 2023-04-28:
Hi @EnthusiasticRoboticist  Do you see a difference in performance if you use Is instead of As in **auto motion = frame.as<rs2::motion_frame>();** like in https://github.com/IntelRealSense/librealsense/issues/4018#issuecomment-520827544

EnthusiasticRoboticist, 2023-04-28:
Hi @MartyG-RealSense, I observe the same behavior. This is the change you expected?

```.cpp
  auto callback = [&](const rs2::frame &frame)
  {
    std::lock_guard<std::mutex> lock(mutex);
    if (frame.is<rs2::frameset>())
    {
      rs2::frameset fs = frame.as<rs2::frameset>();
      std::cout << "nonIMU data" << std::endl;
      // With callbacks, all synchronized stream will arrive in a single frameset
      for (const rs2::frame &f : fs)
        counters[f.get_profile().unique_id()]++;
    }
    else
    {
      std::cout << "gyro: " << (motion.get_profile().stream_type() == RS2_STREAM_GYRO)
                << ", accel: " << (motion.get_profile().stream_type() == RS2_STREAM_ACCEL) << std::endl;
    }
  };
```

MartyG-RealSense, 2023-04-28:
Due to the complexity of the script, it may be worth testing the simpler non-callback example script provided by Intel at the link below to see if it works as expected.

https://github.com/IntelRealSense/librealsense/blob/master/doc/d435i.md#api

EnthusiasticRoboticist, 2023-04-28:
@MartyG-RealSense for non-callback, I don't get anything and `wait_for_frames` always timing out.  I used `rs-callback` [example here](https://github.com/IntelRealSense/librealsense/blob/master/examples/callback/rs-callback.cpp) as starting point of my test code.

EnthusiasticRoboticist, 2023-04-28:
This always times out with `"Frame didn't arrive within 1000"`. I ran it multiple times. No success.

```.cpp
TEST(realsense_test, imu_test_wait_for_frame)
{
  rs2::pipeline pipe;
  rs2::config cfg;
  cfg.enable_stream(RS2_STREAM_ACCEL);
  cfg.enable_stream(RS2_STREAM_GYRO);
  pipe.start(cfg);

  while (true) // Application still alive?
  {
    rs2::frameset frameset = pipe.wait_for_frames(1000);

    if (rs2::motion_frame accel_frame = frameset.first_or_default(RS2_STREAM_ACCEL))
    {
      rs2_vector accel_sample = accel_frame.get_motion_data();
      std::cout << "Accel:" << accel_sample.x << ", " << accel_sample.y << ", " << accel_sample.z << std::endl;
    }

    if (rs2::motion_frame gyro_frame = frameset.first_or_default(RS2_STREAM_GYRO))
    {
      rs2_vector gyro_sample = gyro_frame.get_motion_data();
      std::cout << "Gyro:" << gyro_sample.x << ", " << gyro_sample.y << ", " << gyro_sample.z << std::endl;
    }
  }
}
```

MartyG-RealSense, 2023-04-29:
I looked at your DonkeyJet project and see that it uses both D435i and T265.  

https://www.enthusiasticroboticist.com/blog/introduction-to-my-jetson-based-robot-donkeyjet/

As the T265 is equipped with the same 'Bosch BMI' IMU as the 400 Series cameras, if D435i and T265 are both attached to the Jetson (it looks as though the T265 is rear-mounted) then I wonder whether the program is confused about which of the two IMUs to access.  In Intel's 400 Series and T265 guide **Better Together**, they avoid this scenario by using an IMU-less D415 or D435 model as the depth camera.

https://www.intelrealsense.com/depth-and-tracking-combined-get-started/

[image]

If you do have D435i and T265 active simultaneously, you may be able to resolve the issue by specifying that the accel and gyro streams must be accessed from a camera with a specific device name (such as D435i) via the SDK instruction **RS2_CAMERA_INFO_NAME**.  A C++ script at https://github.com/IntelRealSense/librealsense/issues/2219#issuecomment-412899468 demonstrates this method.

for (auto dev : ctx.query_devices()) // For each device do :
{
if (strcmp(dev.get_info(RS2_CAMERA_INFO_NAME), "Intel RealSense D415") == 0) // Check for compatibility
{

You could alternatively differentiate between the two devices by serial number if you always use the same two cameras.

EnthusiasticRoboticist, 2023-05-01:
I don't have the T265 camera connected in these tests.

MartyG-RealSense, 2023-05-01:
It is recommendable to budget a maximum of 2 watts power draw for each 400 Series camera.  Most computers should be able to handle the power needs of a single camera when the camera is inserted into the computer's USB port.  

If your model of Nano has a **barrel jack** power connector then Intel has Nano-specific advice that strongly recommends enabling it for additional power using the instructions at the link below.

https://jetsonhacks.com/2019/04/10/jetson-nano-use-more-power/

EnthusiasticRoboticist, 2023-05-01:
I am using a 5V/4A supply through the barrel jack.

MartyG-RealSense, 2023-05-02:
Are you using the official 1 meter USB cable supplied with the camera or a longer cable of your own choice, please?

EnthusiasticRoboticist, 2023-05-02:
I am using this cable: https://www.bhphotovideo.com/c/product/1420563-REG/sabrent_cb_c3x1_usb_3_0_to_usb_c.html
which is 0.3 meters.

MartyG-RealSense, 2023-05-02:
Thank you.  Inexpensive sync / charge cables are not ideal for use with RealSense cameras.  A high quality cable designed for data transfer should be used due to the high volume of data bandwidth that RealSense cameras can transmit through the cable.

MartyG-RealSense, 2023-05-08:
Hi @EnthusiasticRoboticist  Do you require further assistance with this case, please?  Thanks!

EnthusiasticRoboticist, 2023-05-08:
Hi Event the original cable had the same issue.

I gave up the D435i. I switched to D435 + T265. It works as expected.

MartyG-RealSense, 2023-05-08:
Thanks very much for the update.

closed 2023-05-12 by EnthusiasticRoboticist