realsenseai/librealsense #11743: System.Exception:“Frame didn't arrive within 5000”,ExternalException: rs2_pipeline_wait_for_frames(pipe:00000201D5C93D20)

975085538, 2023-04-28:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC|
| SDK Version                            |  { 2.0 }                          | 
| Language                            |  {C#}                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I wanted to connect and display the D455 camera depth mode with wpf, but I got an error while running:**Frame didn't arrive within 5000**. **The following is my code snippet:** 
var pipeline = new Pipeline();
            pipeline.Start();
            while (true)
            {
                using (var frames = pipeline.WaitForFrames())
                {
                    using (var depthFrame = frames.DepthFrame)
                    {
                        if (depthFrame != null)
                        {
                            var bytesPerPixel = depthFrame.BitsPerPixel;
                            var pixelData = new byte[depthFrame.DataSize];
                            Marshal.Copy(depthFrame.Data, pixelData, 0, (int)depthFrame.DataSize);
                            var imageWidth = depthFrame.Width;
                            var imageHeight = depthFrame.Height;
                            var bitmap = BitmapSource.Create(imageWidth, imageHeight, 96, 96, PixelFormats.Gray8, null, pixelData, imageWidth * bytesPerPixel);
                            this.image.Source = bitmap;
                        }
                    }
                }
            }
Hope to answer as soon as possible, thank you!

MartyG-RealSense, 2023-04-28:
Using the RealSense SDK's **C# Cookbook** page of example snippets as a reference, it appears that the examples on that page do not have an opening bracket in-between the 'using' statements and instead structure the code like this:

```
using (var frames = pipeline.WaitForFrames())
using (var depthFrame = frames.DepthFrame)
{
```

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/Documentation/cookbook.md

So you could first try removing the opening bracket between the two using statements, and removing one of the four closing brackets - } - at the end of the snippet.

975085538, 2023-04-28:
Thank you very much for your reply. I modified the code according to your suggestion, but an error still occurred: **Frame didn't arrive within 5000.** I don't think it's the parentheses in the code that cause this type of exception. **Here is my revised code:**
var pipeline = new Pipeline();
pipeline.Start();
while (true)
{
              var frames = pipeline.WaitForFrames();
              var depthFrame = frames.DepthFrame;
              if (depthFrame != null)
              {
                  var bytesPerPixel = depthFrame.BitsPerPixel;
                  var pixelData = new byte[depthFrame.DataSize];
                  Marshal.Copy(depthFrame.Data, pixelData, 0, (int)depthFrame.DataSize);
                  var imageWidth = depthFrame.Width;
                  var imageHeight = depthFrame.Height;
                  var bitmap = BitmapSource.Create(imageWidth, imageHeight, 96, 96, PixelFormats.Gray8, null, pixelData, imageWidth * bytesPerPixel);
                  this.image.Source = bitmap;
              }
}
Occurs in a row:var frames = pipeline.WaitForFrames();

MartyG-RealSense, 2023-04-28:
It may be better to continue with **using** - such as `using (var frames = pipeline.WaitForFrames())` - as a 'using' block helps to ensure that frames get garbage collected and the pipeline does not jam up, as advised by a RealSense team member at https://github.com/IntelRealSense/librealsense/issues/2845#issuecomment-444448923 in another case involving the **Frame didn't arrive within 5000** error.

975085538, 2023-05-04:
> It may be better to continue with **using** - such as `using (var frames = pipeline.WaitForFrames())` - as a 'using' block helps to ensure that frames get garbage collected and the pipeline does not jam up, as advised by a RealSense team member at [#2845 (comment)](https://github.com/IntelRealSense/librealsense/issues/2845#issuecomment-444448923) in another case involving the **Frame didn't arrive within 5000** error.

Thank you very much for the problem case you provided. I didn't see the way to solve this problem directly. Could you please give me a detailed answer?

MartyG-RealSense, 2023-05-04:
My knowledge of C# programming is limited, unfortunately.  There is though some example C# code at https://github.com/IntelRealSense/librealsense/issues/3413#issuecomment-474451376 that you could try.

closed 2023-05-10 by 975085538