realsenseai/librealsense #11746: I didn't find the lib file for glfw in your third-party, could you please re-upload a copy?

YANSU7, 2023-04-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2023-04-28:
Hi @YANSU7  Which method did you use to install librealsense, please?

If you used the **installation_md** instructions to build librealsense from source code with CMake then a command in the **Distribution-specific packages** section should be used to install GLFW support.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

[image]

YANSU7, 2023-04-29:
Sorry, I didn't describe my question clearly. I am using the operating system windows 10 and I am going to develop on Visual Studio 2019 platform. The problem I'm having is that I need to add the include and lib folders in order to call your camera while using VS, but after installing your Intel.RealSense.SDK-WIN10-2.49.0.3474.exe I didn't find the corresponding glfw-imgui.lib file, I only I only found the glfw3.h file. Later I downloaded some files like glfw3.lib, glfw3dll.lib, opengl32.lib and so on from the internet and added them to my VS project library directory, and found that the #include <GLFW/glfw3.h> in example.hpp no longer reported an error and the program compiled normally. I tried to change the camera configuration using Intel RealSense Viewer but it still did not work. The requested format is not supported by this demo! So I guess it could be a problem with the glfw3.lib, glfw3dll.lib, opengl32.lib, etc. So I would like to ask you to provide me with your compiled glfw3.lib, glfw3dll.lib, opengl32.lib, etc. I would also like to ask you to provide me with your compiled glfw3.lib, glfw3dll.lib, opengl32.lib, etc. Of course, there may be other problems, so I hope you can help me in some way.
The code I am using is rs-measure.cpp.

MartyG-RealSense, 2023-04-29:
The Windows version of the RealSense SDK provides three **.props** property sheet files for easy linkage of RealSense into your own project using Visual Studio.   A RealSense user created a guide to setting up a Visual Studio project using these props files.  

https://github.com/EduardoWang/visual-studio-real-sense-record-and-playback-project

The prop files are located in the root directory of the RealSense SDK folder at this Windows location after the full SDK has been installed.

**C: > Program Files (x86) > Intel RealSense SDK 2.0**

The guide also provides information about adding glfw3 to your project.

closed 2023-05-12 by MartyG-RealSense